Ultimaker/Cura #20421: 5.10.0 beta 1 changing profile settings

g4dgetboy, 2025-03-20:
### Cura Version

5.10.0 beta 1

### Operating System

Mac os ventura 13.7.4

### Printer

Monoprice select Mini v2

### Reproduction steps

Downloaded and installed beta. Sliced small file. Exported to SD card. Started print on printer and first layer failed due to under extrusion. Checked machine for mechanical issue, none found. Printed previously sliced file without problem. Checked slicer settings to find that all the line width settings are set to 0.18mm and I didn't do it! Checked all the other printer profiles and they are all okay apart from this one printer. Weird :\

### Actual results

as above

### Expected results

as above

### Add your .zip and screenshots here ⬇️

[sight raising block.3mf.zip](https://github.com/user-attachments/files/19375148/sight.raising.block.3mf.zip)

[image]

Frederic98, 2025-03-31:
You've changed the nozzle size to 0.2mm, which causes the line width to be 0.18mm. Does this match your machine? If so, a 0.18mm line width seems fine. If not, set your nozzle size to a value that matches your machine.

g4dgetboy, 2025-03-31:
Hi, thanks for contacting me. I changed nothing on the profiles for this machine. Only this machine was effected and it was all profiles for this one machine. I finished a slice and print, saw the beta was available, installed it and sliced the same file but the line widths on all profiles for this machine regardless of nozzle size were 0.18. None of my other machine profiles had a problem. I've deleted and replaced all the profiles for this machine and  all is back to normal. Sent from my Galaxy