Ultimaker/Cura #20423: Combing going over walls and not using Z-Hop

onolox, 2025-03-21:
### Cura Version

5.9

### Operating System

Windows 10

### Printer

Ender S1

### Reproduction steps

When combing mode is enabled (Any option) it don't let z-hop to occur, it goes straight over printed lines.

### Actual results

It goes over printed parts.

### Expected results

It have to avoid all printed parts of the same layer.

### Add your .zip and screenshots here ⬇️

https://youtu.be/CA3oFaXiyVM
https://youtu.be/E7aRYyvyaL4

[Combing_problem.zip](https://github.com/user-attachments/files/19377460/Combing_problem.zip)

GregValiant, 2025-03-21:
Thanks for the report.
When I slice the project file, there are Z-hops in the gcode.
G1 X109.68 Y102.819 E48.38942
G1 F2400 E47.38942
**G1 F600 Z1.28**
G0 F15000 X101.588 Y107.316 Z1.28
**G1 F600 Z0.28**
G1 F2400 E48.38942
There are a total of 233 Z-hops with your settings.

Playing around with my printer active I don't see any Z-hops in the gcode.  That's really odd because I just did a print with Z-hops and they were fine.  Now I can't generate them.

The Cura team will take a look.

onolox, 2025-03-21:
There are some z-hops here and there, but most of the time when combing is on z-hops are no used, I think because the engine relies on combing and it is faulty, not going around the walls.

GregValiant, 2025-03-21:
I seem to recall you had it set to "within infill" and there wasn't very much.  I generally use "All" but even with that, there weren't a lot of Z-hops.
I wrote this post processor to add z-hops separate from Cura.  It's based solely on the travel distance and has nothing to do with combing or whether there is a retraction or not.  If you stick it into the "scripts" sub-folder of your "Configuration" folder it will be available with the other post processors.  If there are issues let me know.  At some point I will submit it to UM for Cura.
It is best to turn Z-hops off in Cura or there will be some double-z-hops.  That's just slow.
[ZhopOnTravel.zip](https://github.com/user-attachments/files/19397097/ZhopOnTravel.zip)

It looks like this.
[image]

onolox, 2025-03-21:
Combing is broken, this is the real problem, so I will stay with disabling it for now until is corrected, like a peasant.

DasSmeagle, 2025-06-07:
I think I ran into this problem, I get no z hops at all. But toggling Combing does not change that!?