Ultimaker/Cura #20428: Print result is different from view in cura

Eehbahgum, 2025-03-21:
### Cura Version

5.10.0-beta1

### Operating System

Windows 11

### Printer

Creality Ender3 S1 Pro

### Reproduction steps

Opened Cura beta
Loaded file 垃圾袋固定器.stl (https://makerworld.com/en/models/438961-garbage-bag-holder#profileId-344506)
Set the multiple setting to 4

IT WOULD BE USEFUL TO BE ABLE TO UPLOAD THE EXPORTED CURAPROFILE RATHER THAN CONVERTING TO TEXT
[CuraSettings.txt](https://github.com/user-attachments/files/19387978/CuraSettings.txt)

### Actual results

Photos of top and bottom completed view attached as well as the gcode in a zip file as I cannot upload the gcode.

[CuraBug.zip](https://github.com/user-attachments/files/19388193/CuraBug.zip)

### Expected results

I expected to have 4 separate copies with a removable brim. The brim was only added to ensure adhesion.

### Add your .zip and screenshots here ⬇️

[stderr.log](https://github.com/user-attachments/files/19388439/stderr.log)

[cura.log](https://github.com/user-attachments/files/19388444/cura.log)

Added in case they were useful.

GregValiant, 2025-03-21:
Thanks for the report.
I see in the gcode that you have "Spiralize" turned on.  It is impossible to spiralize multiple models "All at Once".  They are separate "islands" on the build plate and there must be travel from island to island.  That's what all those cross-part walls are.
You can print them "One at a Time" in spiralize because that would not require travel between the models.
The extrusions I've highlighted are all caused by the multiple parts and spiralize.
[image]

That looks a lot like the gcode preview in Cura.

[image]

I don't see a bug here.  It appears that it printed the way you set it up.
I'm going to remove the bug label.  Let me know if you feel that is premature.

GregValiant, 2025-03-21:
I've been doing triage for a few years now.  80% of the problems are model problems, 15% of the problems are user problems, and about 5% are actual Cura bugs.  Those bugs tend to get noticed and there are a lot of duplicate reports so it looks like a lot but it isn't really very many.  Some of the ones that are bugs are real beauties and tough to dig out.

GregValiant, 2025-03-22:
I'll go ahead and close this as it isn't a bug.  I hope you understand.

closed 2025-03-22 by GregValiant