Ultimaker/Cura #20429: Cura crash on loading a file

Eehbahgum, 2025-03-21:
### Cura Version

5.10.0-beta1

### Operating System

Windows 11

### Printer

Creality Ender3 S1 Pro

### Reproduction steps

Following on from my report of an unexpected print result, I cleared the plate and reloaded the same file to remove the brim setting and move the copies further apart.

### Actual results

Cura crashed with no error displayed
Reloading Cura, I was able to load the same file successfully.
I selected the model and chose to make 3 copies in a grid pattern without a brim.

### Expected results

Not crashing!
log files attached

[cura.log](https://github.com/user-attachments/files/19388761/cura.log)

[stderr.log](https://github.com/user-attachments/files/19388764/stderr.log)

### Add your .zip and screenshots here ⬇️

Had to print the file using a raft and the attached photo shows what was printed which is not as expected.

[image]

GregValiant, 2025-03-22:
This appears to be a duplicate of #20428 so I'll go ahead and close this.

GregValiant, 2025-03-24:
I guess I was moving too fast and missed that it was a second problem.  I'll go ahead and re-open this.

There are a lot of qml errors in your log file.  This one appears several times.
_"5.10.0-beta.1/share/cura/resources/qml/Settings/SettingView.qml:329:13: QML Connections: Detected function "onSetScrollPositionChangeLoseFocus" in Connections element. This is probably intended to be a signal handler but no signal of the target matches the name."_

Someone from the Cura team will take a look.

Eehbahgum, 2025-03-24:
Easily done, been there and done that :-)Did you want me to append any further incidents or should I raise a new ticket?Get BlueMail for DesktopGregValiant wrote: I guess I was moving too fast and missed that it was a second problem. I'll go ahead and re-open this.
There are a lot of qml errors in your log file. This one appears several times.
"5.10.0-beta.1/share/cura/resources/qml/Settings/SettingView.qml:329:13: QML Connections: Detected function "onSetScrollPositionChangeLoseFocus" in Connections element. This is probably intended to be a signal handler but no signal of the target matches the name."
Someone from the Cura team will take a look.
I guess I was moving too fast and missed that it was a second problem. I'll go ahead and re-open this.
There are a lot of qml errors in your log file. This one appears several times.
"5.10.0-beta.1/share/cura/resources/qml/Settings/SettingView.qml:329:13: QML Connections: Detected function "onSetScrollPositionChangeLoseFocus" in Connections element. This is probably intended to be a signal handler but no signal of the target matches the name."

Eehbahgum, 2025-03-24:
I am crashing the beta every time when I attempt to open an stl file.

[cura.log](https://github.com/user-attachments/files/19433341/cura.log)

Stl file BFTrayBolt v5 has been sliced and printed successfully by cura but now will not load.
V7 of the file has some modifications but will not load.
Cure 5.9.0 loaded and sliced version 7 of the STL file

What is the best way to upload the STL, if you need it?

GregValiant, 2025-03-24:
I don't need a file because I know I can't duplicate this on my Windows 10 system.  I open numerous model and project files from all over the place every day.  I've never had the problem.  Somehow this problem is local to your machine.

You can try renaming the "5.10" folder within in your configuration folder and then un-installing and re-installing Cura 5.10beta.  That will force it to do a clean install with no plugins or options.  If there is an issue with the way the QML controls are loading, or with a plugin, then it should go away.  If there is a conflict between Cura and some other piece of software, then I'm not sure where to go with this.

Does the problem occur when you use 5.9.0 or 5.9.1?

Eehbahgum, 2025-03-24:
My installation of 5.9.0 successfully opens and slices the offending STL file. This has been printed successfully.
I did as you suggested and uninstalled the beta, renaming the folder and then downloaded a fresh copy.
This copy installed and started but again crashed on opening the same STL file.
I restarted Cura again and the file displayed for a short period of time before the program crashed.

[cura.log](https://github.com/user-attachments/files/19437847/cura.log)

I have downloaded 5.9.1 and will update to this version just to check.

HellAholic, 2025-03-25:
The App crash is not registered in the logs and if you don't get a pop up window with the error report when the issue happens, then it's most likely a system crash. [If Cura just suddenly closes and no warnings / messages are visible]

If that is the case, my strongest suspicion would be the drivers, specifically the graphics driver. I have seen few instances of the report so far on windows 11 and 10 after a windows update.

To check, you have two options:

Option A - Update the graphics driver to the latest available (In your case Nvidia), please download the latest from the Nvidia themselves rather than windows or other 3rd party (AIB such as Dell / Lenovo) as their drivers are often not up to date. If the issue is from the driver, then it should resolve after this.

Option B - Open Cura 5.10, load the STL and when the crash happens on your operating system:
- Open Event viewer from the search bar
- Under Event Viewer (Local) > Windows Logs > Applications
- Sort the events by Date and Time and find one with the Source Cura that corresponds to the time of your Crash ( Big red ❌ icon).
- Double-Click on that Event. If you see references to ...\SYSTEM32\ntdll.dll then it's most likely the graphics card driver. (then follow the steps of Option A)
- If there are other messages, you can trouble shoot from there with reason for crash.

Eehbahgum, 2025-03-26:
I have downloaded the latest version of the graphics driver and have installed the release version of Cura 5.10.
I can now open the STL file and slice it :-)
Thank you to you both for taking the time to investigate and resolve the problem

closed 2025-03-26 by Eehbahgum