Ultimaker/Cura #20436: Travel path in Cura creating wrong shapes

onolox, 2025-03-23:
### Cura Version

5.9

### Operating System

Windows 10

### Printer

Ender S1

### Reproduction steps

Trying to print a test I have seen that Cura makes wrong shapes in 2 tests. I compared with Prusa-Slicer, and Prusa prints correctly.

### Actual results

The first photo shows what prusa-slicer generates in the app and when printed.

[image]
[image]

### Expected results

I've made many attempts to get this print to work, but as the results show there's no way to make it print correctly. I'm  posting here some of the attempts.
This shows what Cura produces:

[image]
[image]
[image]
[image]

### Add your .zip and screenshots here ⬇️

The problem is the result of the poor choice of travel path in Cura.

GregValiant, 2025-03-24:
Thanks for the report.
All-in-all, I'd call both the Cura print and the Prusa print unacceptable.  That's a decent test with a 0.4 nozzle, but a torture test with a 0.6 nozzle.

Set Cura up to slice and then use the "File | Save Project" command and then zip the 3mf file and post the zip folder.
Part of this could be settings, but part could also be the printer.
Did you "calibrate the E-steps"?

This is Cura 5.9.0 on an Ender 3 Pro with a 0.6 nozzle.  If I had bothered to tune it for this print it would have been better.  The flow is bouncing up and down and so is the nozzle pressure.  That makes for inconsistent extrusions.

[image]

onolox, 2025-03-24:
[AIl_In_One_3D_printer_test_micro cortado.zip](https://github.com/user-attachments/files/19429126/AIl_In_One_3D_printer_test_micro.cortado.zip)

GregValiant, 2025-03-24:
When you have a model with fine features, printing slower, and at a reduced layer height becomes a necessity.  The pressure changes fluctuate so much in the nozzle that there is no consistency to the extrusions.  In addition, you are printing PLA and your hot end temperature is above the maximum for most PLA manufacturers.  That makes it soupy and prone to oozing.  That's why your print looks so sloppy.  You have combing turned off and so all travel moves are exactly point-to-point with no attempt at avoiding anything.
Your "Flow" numbers are all at 95%.  If your "Steps/mm" were calibrated correctly then your flows would all be at 100%.  Right now you appear to be over-extruding (even at 95% flow).  Some of that over-extrusion could have to do with the hot end temperature but it is also likely that your printer isn't tuned correctly.

- Consider scaling the model up to 200%, or printing this test with a 0.4 nozzle.  I think you would get a much better idea of how to set up for a 0.6 nozzle if the model had larger features.  Large nozzles are for large models with "gross" features.  Small nozzles are for fine work.

If you don't care to do that then:

- Reduce your "Retraction minimum travel" to 0.10 to insure retractions between the fine lines.
- Set the combing mode to "All" and make sure "Avoid printed parts when traveling" is turned on.
- Change the layer height to 0.16 to avoid massive changes in the rate-of-flow through the nozzle.
- Calibrate the E-steps and _DO NOT_ use one of those silly "single wall calibration cubes".  Just make sure that when the printer is told to push 100mm of filament through the extruder, that exactly 100mm is pushed.  Then set all the flows to 100%.
- Drop your print speed to 50 and the outer walls to 35.  This is a calibration test, not a race.  The speed affects how fast the nozzle pressure falls off and has to build back up.  When you are flowing so little through a large nozzle, it becomes important.
- You might want to print a temperature tower.  220 is really hot for PLA.

I think it's printing exactly how you set Cura up.  I don't know how you set up the Prusa slice, but that print isn't very good either.
I agree you aren't getting very good results but I don't see a bug here so I'll remove the label and close this.  I hope you understand.