Ultimaker/Cura #20439: Include "triangle" library in Python environment

Slashee-the-Cow, 2025-03-24:
### Is your feature request related to a problem?

I'm working on a plugin (I've been doing that a lot lately). I tried extruding a shapely Polygon with trimesh.creation.extrudePolygon() but that relies on using an external library (either triangle or mapbox_earcut) to convert the polygon to a translation. Here's the docstring from the trimesh code:
```
def triangulate_polygon(polygon,
                        triangle_args=None,
                        engine=None,
                        **kwargs):
    """
    Given a shapely polygon create a triangulation using a
    python interface to `triangle.c` or mapbox-earcut.
    > pip install triangle
    > pip install mapbox_earcut

    Parameters
    ---------
    polygon : Shapely.geometry.Polygon
        Polygon object to be triangulated
    triangle_args : str or None
        Passed to triangle.triangulate i.e: 'p', 'pq30'
    engine : None or str
      Any value other than 'earcut' will use `triangle`

    Returns
    --------------
    vertices : (n, 2) float
       Points in space
    faces : (n, 3) int
       Index of vertices that make up triangles
    """
```
I would say it seems odd that it includes trimesh without including all of its dependencies, but triangle is only a soft dependency and I'm not sure it ever comes up outside of extruding a polygon, so an edge case that could easily be missed.

### Describe the solution you'd like

Include the "triangle" library in the libraries that ship as part of Cura's Python environment so that it's available to trimesh

### Describe alternatives you've considered

I looked at the possibility of including the code for either of those libraries however they rely on compiled C extensions, so I can't just use the straight code, and I don't know if I can even use a wheel in a plugin, but I'd need to include one for every platform, and I couldn't test macOS (don't have a Mac).
(And obviously I looked at all the primitives trimesh can create and the shape I want isn't one of them)

### Affected users and/or printers

Admittedly "I'm writing a plugin" seems like an incredibly small audience however I would like to use this for a new feature in my "Calibration Shapes Reborn" plugin (currently at 22.6K downloads 😁) so not only could existing users potentially make use of it, but some people may be interested purely for this feature (it's to help test bridging capability).

fieldOfView, 2025-03-24:
Could you use addConvexPolygonExtrusion from Uranium? Certainly more work than "just" using triangle via MeshTools directly, but it would be something that you can use with any version of Cura, not just a future version of Cura that includes the triangle package.
https://github.com/Ultimaker/Uranium/blob/5.9/UM/Mesh/MeshBuilder.py#L790

Slashee-the-Cow, 2025-03-24:
| Wouldn't you need Blender or the Manifold3d library for that?
According to [the docs](https://trimesh.org/trimesh.boolean.html), yes.
But since it works I'm going to assume at least one of those libraries in included in Cura.

fieldOfView, 2025-03-24:
I think it works because you have Blender installed.

Slashee-the-Cow, 2025-03-24:
> I think it works because you have Blender installed.

Well Blender wasn't in my system's PATH but I uninstalled it anyway and it still worked.  
Then I looked at the code for Trimesh and it's a pernicious little sucker. Not only will it use the PATH but it also looks in the default install directories for Blender, OpenSCAD and vhcad (which I've never even heard of) or using the gmsh library if that's installed.  
In my case it was OpenSCAD it was relying on. 
  
So does this mean my dreams of easy boolean operations in Cura are dashed, or at least locked up behind seeing if there's a way to get Trimesh to tell me if there's a backend it can use installed, and if there isn't give the user an error saying "you have to have at least one of these programs installed"?

Also @fieldOfView technically I gotta ask: MeshTools is AGPL licenced. You okay with me taking code from it for something under the GPL?