Ultimaker/Cura #20441: Post processing script At layer change GCO to insert all in uppercase not working with Klipper

bobRoss99999, 2025-03-24:
### Cura Version

5.1.0-Beta 1

### Operating System

Windows 11

### Printer

Ender 3 v3 ke Using Klipper Btt pad 7

### Reproduction steps

1. Added post processing script: Insert at layer change SET_VARIABLE VARIABLE=layer VALUE=Layer+1
2. Sliced and print

### Actual results

Didn't noticed the **Insert at layer change** notice/tool tips saying that **everything will be chaged to upper case**, so:

Klipper config displaying error **VARIABLE must not contain upper case**.

I understand that this error is by design, but is not working with Klipper GCODE command format.

I suggest to add a warning in the Tooltips and changing code to not change everything to Upper case.

### Expected results

Gcode to insert field not to be converted to upper case

### Add your .zip and screenshots here ⬇️

[image]

[image]

bobRoss99999, 2025-03-25:
Adding this  From klipper documentation:

[image]

HellAholic, 2025-03-25:
[image]
You can use the script version below, made the conversion to upper case a checkbox, if you don't select it, the code will be inserted exactly as you type it in.

[InsertAtLayerChange.zip](https://github.com/user-attachments/files/19455334/InsertAtLayerChange.zip)

[image]

https://github.com/Ultimaker/Cura/pull/20444