Ultimaker/Cura #20442: [5.4.0] Problem with rendering preview on MacOS 15.1

artzhookov, 2025-03-25:
### Cura Version

5.4.0

### Operating System

MacOS 15.1

### Printer

Anycubic Kobra Go with klipper

### Reproduction steps

I just sliced ​​the file and got this preview. This happened accidentally on other files as well.

[v4 caliper align.stl.zip](https://github.com/user-attachments/files/19446708/v4.caliper.align.stl.zip)

### Actual results

The model looks wrong, strange lines diverge in all directions:

[image]
[image]

### Expected results

The model looks correct, the current slice layer is displayed

### Add your .zip and screenshots here ⬇️

[AKGO_v4 caliper bleed block.3mf.zip](https://github.com/user-attachments/files/19446722/AKGO_v4.caliper.bleed.block.3mf.zip)

GregValiant, 2025-03-25:
Thanks for the report.
This happens once in a while but no one has been able to duplicate it on purpose.  It will not affect the gcode but it does make the preview hard to see.
Tell us about your computer video system (the type of video card, driver version, etc.).

I call this one the Peter Maxx bug and it gets reported every couple of months so I've seen it a few times.

This one is my personal favorite.

[image]

artzhookov, 2025-03-27:
I hope this helps at least a little. Here is the system information about my macbook:

**Overview:**
```
  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro16,1
  Processor Name:	6-Core Intel Core i7
  Processor Speed:	2.6 GHz
  Number of Processors:	1
  Total Number of Cores:	6
  L2 Cache (per Core):	256 KB
  L3 Cache:	12 MB
  Hyper-Threading Technology:	Enabled
  Memory:	32 GB
  System Firmware Version:	2069.40.2.0.0 (iBridge: 22.16.11072.0.0,0)
  OS Loader Version:	582~2132
```

**Memory Slots:**
```
ECC:	Disabled
Upgradeable Memory:	No

BANK 0/ChannelA-DIMM0:

  Size:	16 GB
  Type:	DDR4
  Speed:	2667 MHz
  Status:	OK
  Manufacturer:	Micron
  Part Number:	MT40A2G8NEA-062E:J

BANK 2/ChannelB-DIMM0:

  Size:	16 GB
  Type:	DDR4
  Speed:	2667 MHz
  Status:	OK
  Manufacturer:	Micron
  Part Number:	MT40A2G8NEA-062E:J
```

**Graphics AMD Radeon Pro 5300M:**
```
  Chipset Model:	AMD Radeon Pro 5300M
  Type:	GPU
  Bus:	PCIe
  PCIe Lane Width:	x16
  VRAM (Total):	4 GB
  Vendor:	AMD (0x1002)
  Device ID:	0x7340
  Revision ID:	0x0043
  ROM Revision:	113-D3220E-190
  VBIOS Version:	113-D32207P1-019
  Option ROM Version:	113-D32207P1-019
  EFI Driver Version:	01.A1.190
  Automatic Graphics Switching:	Supported
  gMux Version:	5.0.0
  Metal Support:	Metal 3
```

**Graphics Intel UHD Graphics 630:**
```
  Chipset Model:	Intel UHD Graphics 630
  Type:	GPU
  Bus:	Built-In
  VRAM (Dynamic, Max):	1536 MB
  Vendor:	Intel
  Device ID:	0x3e9b
  Revision ID:	0x0000
  Automatic Graphics Switching:	Supported
  gMux Version:	5.0.0
  Metal Support:	Metal 3
  Displays:

Colour LCD:
  Display Type:	Built-In Retina LCD
  Resolution:	3072x1920 Retina
  Framebuffer Depth:	24-Bit Colour (ARGB8888)
  Main Display:	Yes
  Mirror:	Off
  Online:	Yes
  Automatically Adjust Brightness:	Yes
  Connection Type:	Internal
```

RobinTail, 2025-07-05:
Latest Cura (5.10.1)

[image]

smartavionics, 2025-07-05:
This little patch to CuraEngine fixed some similar graphical weirdness I was seeing, might be worth trying in UM Cura.
```
diff --git a/src/communication/ArcusCommunication.cpp b/src/communication/ArcusCommunication.cpp
index a36e11c49..d55c2b961 100644
--- a/src/communication/ArcusCommunication.cpp
+++ b/src/communication/ArcusCommunication.cpp
@@ -175,6 +175,8 @@ public:
         line_velocity_data.append(reinterpret_cast<const char*>(line_velocities.data()), line_velocities.size() * sizeof(float));
         line_velocities.clear();
         path_segment->set_line_feedrate(line_velocity_data);
+        _layer_nr = 0;
+        extruder = 0;
     }
 
     /*!
```

GregValiant, 2025-07-05:
Jeez Mark, you're way into the deep end of the pool.  I hate when people remind me how little I know about all this.

@rburema and @HellAholic is this something that might be used for the Peter Maxx bug?

(I'd really hate to see it go away as I truly enjoy the bug reports about it.)

smartavionics, 2025-07-06:
I'm not convinced that patch will help this current problem because I tried slicing the above project without it and didn't see the funky lines appear but the patch was effective in the past at removing some similar artifacts and also graphical weirdness when printing in one-at-a-time mode. YMMV.