Ultimaker/Cura #20448: Note: Alternating Extra Wall May Cause Random Outer Wall Artifacts

Angelic47, 2025-03-26:
### Is your feature request related to a problem?

Yes.  
The **Alternating Extra Wall** feature can lead to extremely poor outer wall quality, making printed models appear to have **random layer artifacts** (commonly mistaken for extrusion inconsistency or ringing).

### Describe the solution you'd like

Please consider adding a **warning note** in the description of the **Alternating Extra Wall** setting, stating that this feature *may cause noticeable layer artifacts on outer walls*, especially when visual quality is a priority.

### Describe alternatives you've considered

Disabling the *Alternating Extra Wall* option entirely.

### Affected users and/or printers

All users who enable the *Alternating Extra Wall* feature,  
**especially those using printer profiles where this option is enabled by default** (Like Voron 2.4).

### Additional information & file uploads

Alright… Let me tell you a bit about my story.

I’ve been a loyal user of Cura, using versions from 4.13 to 5.9, and I print on a **Voron 2.4**.

For a long time, I’ve encountered a strange layer line issue with my 3D prints.  
This issue manifests as random, horizontal texture patterns that look like inconsistent extrusion.
While it doesn't affect the functionality of the models, it severely impacts their appearance.  

This problem has existed since I first assembled my Voron 2.4, and I was never able to resolve it.  
It even **led me on a 3-year** research journey, where I thought about this layer line issue every time I used the printer.  

Back in 2019/2020, before the advent of the Bambu Labs printers, the Voron 2.4 was considered the top-tier 3D printer.  
Even today, the Voron 2.4 remains one of the high-end open-source models available.  

While it is a DIY printer, its design, assembly structure, and software ecosystem (Klipper + Moonraker + Mainsail) are on par with commercial models.  
At that time, the Voron 2.4 was supposed to be one of the "high-end" 3D printers in the market, with quality far surpassing most commercial models.  

**So naturally, I expected its print quality should match or exceed modern commercial printers like Bambu Lab machines.**

But I ran into a wall.

I checked everything:
- Calibrated flow rates
- Swapped out hotends, heat breaks, even extruder gears
- Debugged firmware
- Dried and changed filament
- Replaced belts, motors, adjusted frame tension
- Tuned PA, PID, ran speed/acceleration tests…

**Nothing worked.** The problem remained *identical*, time and time again.

---

Then, by chance, I discovered that some people with Bambu printers also had this problem, *while others did not*.  
The strange thing was, they were all using the same filament, same printer model.  

This led me to believe that the issue was likely a **software-level** problem.  
More specifically, it had nothing to do with the mechanical or electrical components of the printer or its software system.   

I immediately suspected that it was a **slicing parameter issue**.  
I wanted to figure out what went wrong.  

I began to consider each of the slicing parameters (I’ve been using Cura from version 4.x to 5.9.0).  

---

One night, while lying in bed half-awake, I had a sudden inspiration --
**"Alternate Extra Wall."**

I immediately disabled it, ran a test print -- and the result was crystal clear. The issue disappeared.

I later discovered this setting is **enabled by default** in the [voron2_base.def.json](resources/definitions/voron2_base.def.json) printer definition.

**This... was the black swan in the black swan event.**

---

Now you might ask -- how can this setting cause such a severe issue?

Let’s start with Cura’s own description:

> **Alternate Extra Wall**  
> Prints an extra wall at every other layer.  
> This way infill gets caught between these extra walls, resulting in stronger prints.

In other words, if you enable this feature, Cura will insert an additional wall on every other layer,  
interlocking the infill and walls vertically to enhance the overall strength of the model. 

At first glance, this seems like a completely harmless feature.  
Aside from slowing down the print time and using a little more filament, it doesn’t seem to have any downsides.  

However, **the real world is not an ideal world**.
**This is where the problem lies.**

**Not quite.** Because **real-world materials shrink**.

While PLA has minimal shrinkage, it’s not zero. And for modified PLA (PLA+), PETG, or ABS, this shrinkage becomes far more **noticeable**.  

**For ABS materials, the shrinkage issue is even more pronounced.**
Anyone who has printed with ABS knows that if the print environment isn’t sealed and heated, or if the heated bed doesn’t maintain temperatures above 100°C, ABS will warp and detach from the build platform.

Now, back to the problem. *Why does Alternate Extra Wall cause such severe layer lines?*  
**Because it disrupts the consistency of the shrinkage ratio.**  

**Now imagine this:**  
- Even-numbered layers have an *extra wall*.
- Odd-numbered layers don’t.
- This causes *inconsistent shrinkage patterns* layer by layer.

This constant switching causes tiny, random misalignments -- the newly extruded layer rests on a layer that’s already started cooling and shrinking unevenly.  
As printing continues, these inconsistencies **accumulate**, leading to irregular outer wall textures that look like layer artifacts.

They don’t affect mechanical strength or dimensional accuracy -- but they *ruin the surface finish*.

**This is an issue we hadn't considered before.**

---

In hindsight, this seemingly unrelated setting turned out to be the **real cause of years of frustrating layer artifacts** -- totally unrelated to hardware, filament, temperature, or even firmware tuning.

So I printed two identical test cubes -- same model, same printer, same slicer profile -- only difference: one with *Alternate Extra Wall* on (pink), and one with it off (white).  
The results were night and day.

[image]

[image]

[image]

[image]

[image]

I also tested this across different materials, different printers, different settings -- the pattern remained consistent.

Therefore, a feature that seemed completely irrelevant ended up being the culprit for the layer lines.  
This made it **difficult to pinpoint the issue** --

What seemed like a simple layer line problem was actually unrelated to any mechanical, material, speed, or temperature issues.  

---

Interestingly, I found that **Bambu Studio** (based on PrusaSlicer) and **Orca Slicer** also have this setting -- though it appears to be *disabled by default*.  
Of course, I am not a Bambu user, so *I haven’t verified this in detail*.  

Thus, this explains why some people wonder: 
> "Why does my printer show poor wall quality, but others with the same model don’t?"

**The answer lies in shrinkage behavior + Alternate Extra Wall.**

---

So here I am, writing this.  
I’m not sure whether this qualifies strictly as a Feature Request or a Bug Report --  
But I truly hope this post helps **prevent other users** from losing years chasing ghosts like I did.  

I’ve since tested various filaments, parameters, and even printers.  
After printing countless models, the random outer wall layer artifacts never appeared again, validating the correctness of this issue.  

Please consider adding a clear warning about this setting.  
It's a **hidden dragon** that affects the visual quality of prints far more than users may expect.

GregValiant, 2025-03-26:
This is from "fdmprinter.def.json" which is the basis for all the other printer definitions.

"alternate_extra_perimeter":
        {
            "label": "Alternate Extra Wall",
            "description": "Prints an extra wall at every other layer. This way infill gets caught between these extra walls, resulting in stronger prints.",
            "type": "bool",
            _**"default_value": false,**_
            "limit_to_extruder": "infill_extruder_nr",
            "settable_per_mesh": true

This if from the "overrides" section of the Voron "base" definition file.
"_alternate_extra_perimeter": { "**default_value": true** }_,"

It appears that Voron took it upon themselves to enable the setting in their definition file.
So this would seem to be a Voron issue.  UltiMaker takes the definition files in good faith that the authors got it right.

Angelic47, 2025-03-27:
Thank you for your response.  
I've looked into this more deeply and realized that this issue doesn't just exist in Voron -- it also appears in several other printer definition files.

Through some searching, I found that the following profiles have `alternate_extra_perimeter => True` set:

- [ankermake_m5c.def.json](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/definitions/ankermake_m5c.def.json#L39)
- [atmat_signal_pro_base.def.json](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/definitions/atmat_signal_pro_base.def.json#L48)
- [blocks_base.def.json](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/definitions/blocks_base.def.json#L28)
- [gutenberg_base.def.json](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/definitions/gutenberg_base.def.json#L26)
- [voron2_base.def.json](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/definitions/voron2_base.def.json#L31)
- [vzbot_base.def.json](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/definitions/vzbot_base.def.json#L77)
- [snakeoil_standard_base.def.json](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/definitions/snakeoil_standard_base.def.json#L26)

Voron is just one among them.

Additionally, there are also some quality-related configuration files that include it:

- [resources/quality/eryone_thinker/eryone_thinker_fine.inst.cfg](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/quality/eryone_thinker/eryone_thinker_fine.inst.cfg#L14)
- [resources/quality/eryone_thinker/eryone_thinker_high.inst.cfg](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/quality/eryone_thinker/eryone_thinker_high.inst.cfg#L14)
- [resources/quality/gutenberg/gutenberg_global_normal_quality.inst.cfg](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/quality/gutenberg/gutenberg_global_normal_quality.inst.cfg#L21)
- [resources/quality/gutenberg/gutenberg_global_strong_quality.inst.cfg](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/quality/gutenberg/gutenberg_global_strong_quality.inst.cfg#L21)
- [resources/quality/inat/inat_base_strong.inst.cfg](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/quality/inat/inat_base_strong.inst.cfg#L15)
- [resources/quality/snakeoilxy/snakeoilxy_global_normal_quality.inst.cfg](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/quality/snakeoilxy/snakeoilxy_global_normal_quality.inst.cfg#L21)
- [resources/quality/snakeoilxy/snakeoilxy_global_strong_quality.inst.cfg](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/quality/snakeoilxy/snakeoilxy_global_strong_quality.inst.cfg#L21)

What makes things worse is that `alternate_extra_perimeter` is categorized as an `expert-level` setting:  
- [resources/setting_visibility/expert.cfg](https://github.com/Ultimaker/Cura/blob/fa1ead30647159f4bf1f7f89394f7b937ce21f73/resources/setting_visibility/expert.cfg#L37)
This means the setting is hidden in Cura's UI by default -- users need to manually enable Expert Mode to even see it.

This is almost a textbook example of a "*low-visibility configuration trap.*"  
In reality, most users have no idea that this setting even exists. And among those few expert users who do see it, many probably don't realize how much it can affect the surface quality of the outer walls.

Considering that **almost no community post or official documentation** mentions how *Alternate Extra Wall* can influence layer artifacts, this **can easily mislead users into thinking they're facing common hardware issues**.  
They might then waste a lot of time and effort debugging, only to never find the real cause -- and ultimately, resign themselves to poor surface finish.

*So... Perhaps making this setting visible by default, and adding a warning when the user enables it (or simply including a warning in the setting's description), could be a helpful solution... ?*

As for Voron, I'll try reaching out to their team later to see if they would consider changing the default value of this setting.

GregValiant, 2025-03-27:
The UltiMaker Company cannot be responsible for the setting configurations of 3rd party printers.  I have used that setting on my own prints (Ender 3 Pro) when strength was a priority and I never saw a problem with the finish on the parts.  My case may have been different as it was already a 3 wall part so the 4th wall would not be expected to have an effect.
In the several years I've been doing the triage here this is the first mention I've seen regarding the setting.  My guess would be that it is not a widespread problem.