Ultimaker/Cura #20449: crashes on single lager GCODE when trying to step through files/

glasscanvas, 2025-03-26:
### Cura Version

5.10.0

### Operating System

Mac OS 15.3.2  24D81, M2 AirBook

### Printer

GRBL pen plotter custom 

### Reproduction steps

1. Load GCODE
2. try to scroll through layer or play it - crashes immediatelly 

### Actual results

crashes immediately have to force quit. Earlier version 5.6 had no issues displaying and scrolling through file. This is a one layer GCODE file for a GRBL pen plotter.

### Expected results

preview GCODE layer one motion at a time.

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/user-attachments/files/19473158/cura.log)

[IMG_1485_20250326_130504_cura.gcode.zip](https://github.com/user-attachments/files/19473165/IMG_1485_20250326_130504_cura.gcode.zip)

GregValiant, 2025-03-26:
Thanks for the report.
I'm not sure what to make of this.  When I load that gcode file - Cura only shows travel moves in the gcode preview and they are below the build plate.
Looking at the gcode that seems to match up.  All the Z's are either -5.0 or -15.0.  Since there are no extrusions in the file, all that is left is the travel moves.

When I hit the "playback" button Cura locks up.  From the log file is _"Min thickness can't be calculated because all the values are zero"_ so it might be a "divide by zero" problem.  Whether it is or not, nothing would be shown because there are no values to show.

When I opened the gcode in Prusaslicer preview the same thing happens.  There really is no view at all.

I'll leave the bug label on this because even though I don't expect it will ever show your gcode - it shouldn't crash.
I think you're going to have to find a gcode viewer that can deal with your sorts of files.  You may end up with a CNC machine Gcode viewer as they can dip into the negative Z.
This is very much a niche problem so I would not expect this to get much of a priority.
BTW I couldn't get this to work with 5.6.0 either.

glasscanvas, 2025-03-27:
Greg, thanks for looking at the issue with Cura. This is a Pen plotter GCODE file for a CNC controller with GRBL on an Arduino. It is a one layer file. I built this machine. The pen values are negative. 0 is home,  -5 corresponds to pen up and -15 to pen down (it can go as low as -35.  Prusa shows it as travel and Cura should also display it as travel. Universal GCODE Sender (USG) does display it but I was hoping to use Cura's better display because I want to scroll the plot buildup with the horizontal slider so I can preview it.  USG does not do this. The file does plot/paint correctly to my printer including dipping the brush into several color wells. 

Example of a different file printing:
[https://youtu.be/8XWwEHxJTIg?si=1PjnQBHHkE9yddEy](url)

GregValiant, 2025-03-28:
I took your Gcode file and added lines using the "; START Brush Y Yellow Paint Dip Action" and adding ;LAYER:x lines.  That allowed the simulation to work OK.
It ends up with 256 layers which are superimposed on the same Z heights.

; START Brush Y Yellow Paint Dip Action
;LAYER:1

closed 2025-03-28 by GregValiant