Ultimaker/Cura #20454: 5.10. crash when skipping signin on first run

ErwinBaeyens, 2025-03-27:
### Cura Version

5.10.0 

### Operating System

Fedora Linux 40 Cinnamon spin

### Printer

Sovol SV07 klipper

### Reproduction steps

1. back-up existing profile under $HOME/.config/cura/
2.  start  cura
3. navigate the welcome wizzard 
4. When asked to sign in to the UltiMaker platform click skip

### Actual results

The application quits printing  this:
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
file:///tmp/.mount_curae5MMfd/share/cura/resources/qml/WelcomePages/PrinterCard.qml:63:9: QML ColumnLayout: Binding loop detected for property "height"
Segmentation fault (core dumped)

### Expected results

A working app

### Add your .zip and screenshots here ⬇️

I have no screenshots to add

ErwinBaeyens, 2025-03-27:
Tried again with signing in and the result was the same:
this is what is printed on the console:

qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_UltiMa7n2fTK/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMa7n2fTK/share/cura/resources/qml/Actions.qml"
qt.qml.context: file:///tmp/.mount_UltiMa7n2fTK/share/cura/resources/qml/Settings/SettingView.qml:432:16 Parameter "index" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::startTimer: Timers can only be used with threads started with QThread
QObject::killTimer: Timers cannot be stopped from another thread
qt.qml.context: file:///tmp/.mount_UltiMa7n2fTK/share/cura/resources/qml/WelcomePages/CloudContent.qml:22:5 Parameter "newCloudPrintersDetected" is not declared. Injection of parameters into signal handlers is deprecated. Use JavaScript functions with formal parameters instead.
file:///tmp/.mount_UltiMa7n2fTK/share/cura/resources/qml/WelcomePages/PrinterCard.qml:64:9: QML ColumnLayout: Binding loop detected for property "width"
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
Qt Quick Layouts: Detected recursive rearrange. Aborting after two iterations.
file:///tmp/.mount_UltiMa7n2fTK/share/cura/resources/qml/WelcomePages/PrinterCard.qml:63:9: QML ColumnLayout: Binding loop detected for property "height"
Segmentation fault (core dumped)

caldir65, 2025-03-28:
Same here with (K)Ubuntu 24.04 / X11 in Use / Wayland just installed
Also with and without login

kupher, 2025-04-01:
I had the same issue on arch Linux, here is a fix I did: Restart your DE, Make sure you change it to x11 session. login, open Cura in the x11 session. go through all the promps, also setup your printer. Close cura, restart your DE back and switch back to wayland.

ErwinBaeyens, 2025-04-01:
That could be a solution but as i said above I'm running the Fedora Cinnamon spin which is still X11 not wayland

caldir65, 2025-04-01:
Sorry, but in my case the machine always running x11, and I've no chance to doch a setup during start of cura.
Even I do a setup with an elder version which run and than do an update will not work for c5.10 (last successfull Version which still works I tested ist 5.6...)

ErwinBaeyens, 2025-04-01:
That is similar to the experience I have,   I was not able to run the config wizard on 5.8 either. or on 5.7 It does work on 5.6 
I found that this is somewhat similar to issue #18843 
 but the proposed workaround  :
`./Ultimaker-Cura-5.10.0-linux-X64.AppImage --platformtheme gtk3`
does not seem to work

j8spencer, 2025-05-25:
I see the same behavior on Linux Mint 22.1 Cinnamon with 5.10 and 5.9.1. I entered duplicate bug [20623](https://github.com/Ultimaker/Cura/issues/20623) with logs and details.

The `--platformtheme gtk3` flag also did not help me.

OpenLogicEFI, 2025-07-10:
### Crash after Login – Cura 5.10.1 (Flatpak & AppImage)

**Disclaimer:** I had ChatGPT help me write this to make sure everything is worded clearly and that I provide the most relevant info — I’m not an expert. But all logs and outputs are real and from my system. Hope this helps!  
\- *Aaron*

---
Same issue for me too. Cura 5.10.1 **crashes immediately after login** via browser (OAuth). The app starts fine, opens the browser (Firefox), I log in, and as soon as the browser hands back control, Cura segfaults. This happens with both:

- AppImage
- Flatpak

No crash occurs until the browser login completes. The crash seems related to the **login callback/token handling**, not graphics or sandboxing.

---

###  Repro Steps
1. Launch Cura (`AppImage` or `Flatpak`)
2. Start login process via browser
3. Complete login successfully
4. Cura crashes the moment the tab closes

---

### System Info
<details>
<summary>Click to expand</summary>
OS: Linux Mint 22.1 Xia (Ubuntu 24.04 base)
Kernel: 6.8.0-63-generic
Desktop: Cinnamon 6.4.8 (X11)
CPU: AMD Ryzen 5 9600X
GPU: AMD Radeon RX 9060 XT (driver: amdgpu, radeonsi, mesa 25.0.0)
Monitors: 3x 2560x1440 (DP + HDMI + HDMI)
AppImage: UltiMaker-Cura-5.10.1-linux-X64.AppImage
Flatpak: Latest from Flathub

</details>

---

### Terminal Output Before Crash
```text
Gtk-Message: Failed to load module "xapp-gtk3-module"
qt.qml.typeresolution.cycle: Cyclic dependency detected...
QObject::startTimer: Timers can only be used with threads started with QThread
QML ColumnLayout: Binding loop detected for property "width"
Qt Quick Layouts: Detected recursive rearrange. Aborting...
Segmentation fault (core dumped)

Backtrace from coredumpctl gdb

Core was generated by `/app/cura/UltiMaker-Cura -platformtheme gtk3'.
Program terminated with signal SIGSEGV, Segmentation fault.

#0  0x0000799f7e6a3eb4 in ?? ()
#1  0x0000799f7e381320 in ?? ()
#2  0x7bc6a8f1c4c89d00 in ?? ()
#3  0x322d677673726269 in ?? ()
#4  0x000000000000000b in ?? ()
#5  0x0000000000000000 in ?? ()
```
---
Notes
    Same crash behavior in both AppImage and Flatpak builds
    Happens only after login completes
    Tried the following with no success:
1. --platformtheme gtk3
2. --single-instance
3. LIBGL_ALWAYS_SOFTWARE=1
4. Switching default browser to Chromium
5. Launching with full Flatseal access in Flatpak

Nothing relevant in logs before the crash except QML layout issues and segfault

---

Let me know if you'd like me to test a debug build or try other troubleshooting steps.
-Aaron