Ultimaker/Cura #20459: Cura is generating odd column support structure

doc-moss, 2025-03-29:
### Cura Version

5.10

### Operating System

macOS Sequoia Version 15.3.2

### Printer

Ultimaker S3

### Reproduction steps

**Problem:**
When enabling supports on Cura, cura generates a ring structure on the top right hand corner of the build plate. I do not think this is expected behavior and my troubleshooting and google-ing have not turned up any results. 

**What have I done to trouble shoot**
I have confirmed there is no obj data that is exported in that corner.  This happens no matter what I load to Cura (i.e. different models, different projects), the generate object is not selectable and cannot be moved when "all models" are selected. The generated model appears to always appear in the same place. I have attempted to install both the ARM and x64 version of Cura (on mac os). I have attempted importing different file types. The odd support structure only appears after I enable supports (disappears if I disable supports) 

**Finally, how to reproduce**
Just import anything into cura (note, all my files were generated with Autodesk Fusion

### Actual results

Cura generates an odd ring structure in the right hand corner (always in this location) 

### Expected results

It should not generate this structure

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[image]
[image]

Here are the project/log files ( I included the original model file too)
[Project Files.zip](https://github.com/user-attachments/files/19523110/Project.Files.zip)

Asterchades, 2025-03-30:
That isn't a Support - that's a Prime Tower. You can adjust its settings, including disabling it, in the _Dual Extrusions_ settings section.

doc-moss, 2025-03-30:
Of course it was user error haha, I had a sneaking suspicion given that I could not find other issues online. Thanks.

closed 2025-03-30 by doc-moss