Ultimaker/Cura #20460: Version 5.10.0 Crashes when opening Project and GCODE files

plainpylut, 2025-03-30:
### Cura Version

5.10.0

### Operating System

Windows 10

### Printer

Creality Ender 5 Plus

### Reproduction steps

1.  Install Cura 5.10.0 along with Cura 5.9.1 (default subdirs).
2.  Use attached flies and using version 5.9.1 verify you can load the files without Cura crashing:
     A.  Open: Consendo.3MF.
     B.  Clear build plate
     C.  Open: Consendo.gcode.
     D.  Clear build plate
     E. Open: Consendo.stl.
3.  Repeat steps A through E using Version 5.10.0.  It crashes pretty reliably!
4. Repeat 2A through 2E using WingTipSkid

[Crash.zip](https://github.com/user-attachments/files/19527393/Crash.zip)

.* files.
5. I have check my STL files using MSBuild - No Errors and water tight.
6. I have removed and re-installed 5.10.0 and it still crashes.

### Actual results

Test using attached Crash.zip.

### Expected results

Cura should be able to load 3MF, GCODE, and STL files.

### Add your .zip and screenshots here ⬇️

[Crash.zip](https://github.com/user-attachments/files/19527395/Crash.zip)

GregValiant, 2025-03-30:
Thanks for the report.
I've had this crash occur occasionally going back a few versions and that includes 5.9.0.  I'm also on a Win10 system.
Even with your files I can't get it to crash reliably.  When Cura crashes I haven't seen any notations in the log file either.

A second bad thing that can happen is that the "Slice" button will be non-functional (stuck on "Save to Disk") on a model file opened _AFTER_ a gcode file was dismissed.

There is always a Gcode file involved, but the other file can be any model file.

The Cura team will take a look.

HenraL, 2025-03-31:
# Cura File Importing Issues (Multiple Versions)

## Summary
I'm also experiencing crashes when drag-and-dropping 3mf files into Cura across multiple versions (5.7.1, 5.9.0, 5.10.0) on Ubuntu 24.04.2 LTS.

The verison of cura used are the .AppImage files.

## System Information
- **OS**: Ubuntu 24.04.2 LTS
- **Cura Versions Tested**: 5.7.1, 5.9.0, 5.10.0
- **Hardware Resources**: 14/31.2GB RAM used, 8 cores with minimal load, no swap in use

## Issues By File Type

### 3MF Files
- **All Versions (5.7.1, 5.9.0, 5.10.0)**: Immediate crash on drag-and-drop
- **Version 5.10.0**: Initially offered to "import or open the project" but still crashed; now instantly crashes

### STL Files
- **All Versions**: Very slow to open (almost as long as opening gcode files)
- **Version 5.7.1**: ~2 minutes on Linux vs. seconds on Windows
- **Note**: Some caching effect noticed when opening/closing Cura multiple times

### GCODE Files
- **All Versions**: Extremely slow to open but will eventually render if you wait
- **Version 5.10.0**: Takes about the same amount of time as 5.9 and 5.7

## Error Logs

### Version 5.10.0
**Startup Output**:
```bash
UM/Settings/SettingFunction.py:244: DeprecationWarning: ast.Str is deprecated and will be removed in Python 3.14; use ast.Constant instead
  def visit_Str(self, node: ast.Str) -> None:
Gtk-Message: 02:01:59.654: Failed to load module "canberra-gtk-module"
Gtk-Message: 02:01:59.655: Failed to load module "canberra-gtk-module"
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_curaN582Y1/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_curaN582Y1/share/cura/resources/qml/Actions.qml"
```

**After 3MF Drag-and-Drop**:
```bash
malloc(): smallbin double linked list corrupted
Abandon (core dumped)
```

### Version 5.9.0
**Startup Output**:
```bash
Gtk-Message: 02:08:38.514: Failed to load module "canberra-gtk-module"
Gtk-Message: 02:08:38.515: Failed to load module "canberra-gtk-module"
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_cura5_WP1EOK/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_cura5_WP1EOK/share/cura/resources/qml/Actions.qml"

** (UltiMaker-Cura:542030): WARNING **: 02:09:05.110: atk-bridge: get_device_events_reply: unknown signature
```

**After 3MF Drag-and-Drop**:
```bash
Erreur de segmentation (core dumped)
```

### Version 5.7.1
**Startup Output**:
```bash
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_cura5_RPhc78/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_cura5_RPhc78/share/cura/resources/qml/Actions.qml"
```

**After 3MF Drag-and-Drop**:
```bash
** (UltiMaker-Cura:541742): WARNING **: 02:08:48.337: atk-bridge: get_device_events_reply: unknown signature
QFSFileEngine::open: No file name specified
[... 37 lines cut ...]

logging.basicConfig(
    format='[%(asctime)s] %(levelname)s %(pathname)s:%(lineno)d : %(message)s',
    datefmt='%Y-%m-%d %H:%M:%S',
    level=logging.DEBUG
)
```

Here are some libraries that can provide simillar functionnality for other langages:
* Java: Log4j/Logback
* C: Zlog
* Cpp: Log4cpp
* QML: it is native

## Notes
P.S.: My system is configured to French, hence, the sentence "Erreur de segmentation (core dumped)" translates to "Error segmentation fault (core dumped)". Quite common in C (generally because the program attempted an illegal manipulation of memory).

P.P.S.: I do not know the code, I am talking based off the outputs as well as the few years I have spent programming in C and Python. This is why I am trying to be as factual as possible.

P.P.P.S.: All the tests were drag and drop tests, if tests made via pop-up box (Cura's file explorer) were made it will be explicitly mentioned.

P.P.P.P.S. : sorry for the long post, I just thought it was important to provide as many details as possible when trying to debug an issue.

## Conclusion
These crashes with 3MF files and performance issues with STL/GCODE files appear consistent across multiple Cura versions on Ubuntu 24.04.2 LTS. The cyclic dependency warnings and memory corruption errors suggest there might be underlying issues with QML dependencies or memory management. I hope this detailed report helps in identifying and resolving these issues. I'm happy to provide additional information or testing if needed.

GregValiant, 2025-03-31:
The only thing showing up in my log is:
"2025-03-31 07:46:42,966 - WARNING - [JobQueueWorker [3]] UM.Scene.SceneNode.addDecorator [234]: Unable to add the same decorator type (<class 'cura.Scene.SliceableObjectDecorator.SliceableObjectDecorator'>) to a SceneNode twice."

HenraL, 2025-03-31:
Is there anything I can do to provide more information that would help you?
I still have the 3 different versions.

boomsya, 2025-04-02:
I have crash after slicing start.
5.9.1 works fine