Ultimaker/Cura #20462: Add "outer" Z seam positions

ldenisey, 2025-03-31:
### Is your feature request related to a problem?

Most of my prints are symmetrical. It would be very useful to have a way to set Z seam positions to "outer" values, i.e. two sides positioning. 

To illustrate and clarify my wording, if I choose "front", the positioning is done in the "inner" sides of my part : 

[image]

If I choose " front left", I get an asymmetrical situation, the positioning is "inner" for the right side of the part and "outer" (= left side here) for the left side of the part.

[image]

If I choose right, I have the mirrored situation : 

[image]

### Describe the solution you'd like

It would be great to be able to set a "front outer" positioning that would target both "outer" sides, in those areas : 

[image]

Of course, the idea should be extended to "back outer", "right outer", "left outer" and maybe "outer", which would be a 4 sides positioning.

### Describe alternatives you've considered

As now, I don't a work around. Something by moving and rotating the part of the plate, I can get a not so bad positioning but it is never really satisfying.
Most of the time, I choose front and try and fix that post print with a cutter but it is never very clean.

### Affected users and/or printers

All symmetrical prints might benefit from that.

GregValiant, 2025-03-31:
The "Front, Back, Left, Right..." terminology is just there to help us understand...it's the numbers in the "Z-Seam X" and "Z-Seam Y" boxes that matter.
This problem always comes up when a model has islands develop.  Each island must have a Z-seam.
You can manually enter the Z-seam location into the boxes.  If the coordinates you enter happen to be the midpoint of the build plate, and the model is sitting at the midpoint of the build plate, you can get different effects.
Here I've set the Z-seam to X115, Y115 which is my build plate midpoint.
[image]

I suppose having "Middle" along with the other word options would be of some value.

ldenisey, 2025-03-31:
Thank you for your response.

Indeed, my issue was that a single Z-seam can be an issue when there are several islands.

You have pointed me to a solution. I can specify different Z-seam for each island using blockers, like so : 

[image]

closed 2025-03-31 by ldenisey