Ultimaker/Cura #20464: Not slicing

Oddverse2009, 2025-03-31:
### Cura Version

5.4.0

### Operating System

windows 10

### Printer

Hephi3D Tina2S

### Reproduction steps

selected file withered foxy body from thingiverse
tried to slice wouldn't work

### Actual results

Says sliceing failed due to an unexpected error

### Expected results

it shouldn't slice

### Add your .zip and screenshots here ⬇️

[](url)https://www.thingiverse.com/thing:6922692/files

GregValiant, 2025-03-31:
Thanks for the report.
We need a project file and your cura.log file to investigate this.

Oddverse2009, 2025-04-01:
and i switched to Cura 5.10.0 and that seemed to fix the problem the only reason I was using 5.3.0 in the first place was because the download file came with my 3d-printer

GregValiant, 2025-04-01:
5.3 thru 5.6 had problems with certain types of models.  That finally got fixed in 5.7.0.

I'll go ahead and close this then.

closed 2025-04-01 by GregValiant