Ultimaker/Cura #20465: It seems that the gcode files cut in the current version of the slicer do not provide information about the required nozzle temperature. [5.10]

Stiven-Melodic, 2025-04-01:
### Cura Version

5.10

### Operating System

Windows 10

### Printer

Kingroon kp3s

### Reproduction steps

When loading files sliced ​​in the current version of the file, the printer does not see the specified temperature for the nozzle, but correctly reads the specified table temperature. I checked several files sliced ​​in older versions of the slicer, there are no problems with them and the printer correctly receives the nozzle temperature from the files.

[cap.zip](https://github.com/user-attachments/files/19558355/cap.zip)

### Actual results

I use ABS+ plastic. After starting to print a file on the printer, it starts to warm up as expected. In my settings, 88 degrees Celsius is set for the table heating temperature and 258 degrees Celsius for heating the nozzle. The table, as expected, displays 88/88 degrees with a calibration of 1 degree, but the nozzle displays 0/258. At the same time, if you manually raise the nozzle temperature to the required level in the printer settings, then printing starts.

### Expected results

In the slicer interface, the temperature is displayed correctly, but for some reason the printer reads it as a value of 0, but models cut in older versions of the slicer do not have this problem and are correctly read by the printer.

### Add your .zip and screenshots here ⬇️

.

GregValiant, 2025-04-02:
Thanks for the report.
I sliced your project file and the gcode is correct.
```
Generated with Cura_SteamEngine 5.10.0
M140 S88
M105
M190 S88
M104 S258
M105
M109 S258
G28 ; home all axes <<<<<<<<First line of the startup gcode
```
Since you are not calling out the temperatures in your StartUp Gode, Cura is adding the temperature lines.  That is the normal situation and is expected behavior given your settings.

If the printer is ignoring the Gcode commands then that is a printer problem or maybe a problem with the SD card or a print server (if you are using one).  The gcode is absolutely correct.

I'm going to remove the bug label as this doesn't appear to be a Cura bug.  I'll leave this report open for a bit so you can respond.

Stiven-Melodic, 2025-04-04:
It looks like it was a firmware error. Sorry for the false appeal. Of
course, the error note I submitted should be deleted so as not to mislead.