Ultimaker/Cura #20466: Arachne ignoring "Flow Rate Compensation Max Extrusion Offset" limit for negative offsets

PCChazter, 2025-04-01:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Modded Ender 3

### Reproduction steps

1. Select a profile (I selected Super Quality to reproduce)
2. Set both Minimum Odd/Even Wall Line Width to 0
3. Adjust other settings, results vary depending on other settings and model. These are what were repeatable for me:
- Layer Height 0.16
- Initial Layer Height 0.2
- Line Width 0.45
- Minimum Wall Line Width 0.42
- Wall Line Count 3
- Horizontal Expansion 0
- Initial Horizontal Expansion 0
4. Slice model
5. Search with text editor for "FLOW_RATE_COMPENSATED_OFFSET"

### Actual results

There will be some negative flow rate compensation, even with "Flow Rate Compensation Max Extrusion Offset" set to 0. For me, I noticed this because Klipper errored out with a "Move exceeds maximum extrusion" error. Looking into the GCode, it had an extrusion offset of -1.032192 in the spot that it errored out on, which caused it to extrude 1.04193mm of filament on a 0.3mm movement.

### Expected results

Negative values of "FLOW_RATE_COMPENSATED_OFFSET" should be limited by "Flow Rate Compensation Max Extrusion Offset"

### Add your .zip and screenshots here ⬇️

It seems as though lots of other settings have an effect on how this issue presents itself, as well as the model used. I wanted to effectively turn Arachne off, which settings Minimum Odd/Even Wall Line Width to 0 seemed to work well enough for what I was printing, and I was able to get a couple files to print with those settings until I got this error
[image]

Looking in the GCode at the spot that it failed I found this
[image]

I was able to reproduce it with a Benchy model to try and narrow down settings that cause it.

Project file with Benchy that produced error
[3DBenchy_E3_BADPF_PLA_2025-04-01.zip](https://github.com/user-attachments/files/19558505/3DBenchy_E3_BADPF_PLA_2025-04-01.zip)

These are the settings that I was able to get a benchy to have the negative offsets
[image]

GCode where negative offset is
[image]

GregValiant, 2025-04-02:
Thanks for the report.
When I slice the project I see this in the gcode:
```
G1 F2400 E2039.93892
;FLOW_RATE_COMPENSATED_OFFSET = -0.080000
G1 F3000 X87.508 Y97.748 E2039.85862
```
So yes, the extruder was actually retracting in what would normally be an extrusion move.

Setting the "Minimum" Even and Odd wall line widths to zero could be causing a "divide by 0" error that isn't caught.  
As a workaround for the problem, don't set the minimum line widths to 0.

Someone from the Cura team will take a look.

GregValiant, 2025-04-03:
An 'Arachne' switch has been asked for previously.  I just use 4.13.1 (or the SmartAvionics fork) when the variable line width gets in the way.  As I've gotten used to dealing with Arachne I found that I can dial it in like I could without it.