Ultimaker/Cura #20474: Variable speed although all speeds are set to the same value

stang1966gt, 2025-04-03:
### Cura Version

5.9.0

### Operating System

Mac OS Sonoma 14.4.1

### Printer

CR-10 Klipper

### Reproduction steps

1) CR-10 LEG STL Set up the print speeds to 100mms, sliced and printed
2) Set All print speeds to 100mms, Result: reviewed Preview tab with no change
3) Tested the issue with FlashForged Cube, Result: this time noted changing the bottom layer from 30 to 100 would get rid of the outer layer line speeds.  But this doesn't work for the LEG STL

Minimum timer per layer = 1 or 10 sec, the result is the same.  
# of Slow layer = 1 or 0, same results.
All the speeds are set to 100mm/s
Cooling - Minimum Speed = 100
Overhang Speed = 100
Bridging Off 
Flow Rate Comp factor = 100%
Post Processing - set all layers to 100, still shows as variable but not sure if this would actually change the preview tab view.
All Settings turned ON, I don't see another option to change
Cura 5.9.0

### Actual results

The preview tab shows different speeds on different parts of the STL, despite everything set to 100mms.  The changes in speed can relate to 2 issues: 1) layer smoothness 2) changes in vibration frequencies, faster accelerations and speeds = smoother walls per my input shaping testing.  Please note I changed the Leg STL to remove the roof at the base in an attempt to see if the speed would be more consistent, so the print photo doesn't match the Preview tab exactly.  I don't have the older STL but it had lines that matched with the part, similar to the Cube.

### Expected results

All exterior walls to print at 100mms.

### Add your .zip and screenshots here ⬇️

[Archive.zip](https://github.com/user-attachments/files/19589139/Archive.zip)

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

GregValiant, 2025-04-03:
Thanks for the report.
Those speed variations are because of the way you have Cura set up.  It's doing exactly what you asked for.
2 slower first layers.
Top - Bottom speed at 30mm/sec
Flow Equalization Ration 100%
Flow Rate Compensation Factor 80%
Gradual Flow - Enabled
[image]

With just a couple of settings changes - all speeds are at 100mm/sec.
0 slower first layers.
Top - Bottom speed at 100mm/sec
Flow Equalization Ration 0%
Flow Rate Compensation Factor 0%
Gradual Flow - Disabled
[image]

And here is the other one.
[image]

I'll go ahead and remove the bug label and close this as it is a setting issue.

closed 2025-04-03 by GregValiant

stang1966gt, 2025-04-03:
Thank youSeems Gradual Flow was the winner of the bunchI updated the others too

image.png (view on web)
image.png (view on web)
image.png (view on web)

image.png (view on web)
image.png (view on web)
image.png (view on web)