Ultimaker/Cura #20477: Bug with Extruder activation with Snalmaker when sliced in Cura

B5TRK, 2025-04-03:
### Cura Version

5.10

### Operating System

Win 11

### Printer

snapmaker

### Reproduction steps

When I slice an object in CuraV5.10, even if I choose the right extruder, it defaults to the left extruder. The right extruder works perfedtly on objects sliced in V5.91 or earlier. I have reloaded PC, reloaded Cura 5.10, reset the printer. 

### Actual results

Nothing, it still defaults to the left extruder, even though I choose the right extruder

### Expected results

If I choose the right extruder, it should print on the left extruder

### Add your .zip and screenshots here ⬇️

cannot upload stl

GregValiant, 2025-04-04:
Thanks for the report.
Set Cura up to slice and then use the "File | Save Project" command.  Zip the 3mf file and post the zip folder here.

Do you have the "Snapmaker Plugin" installed?  Cura 5.10 moved to Python 3.12 (I think) and a lot of the plugins needed work in order to be compatible.  In this case the author of the plugin (and your definition file) is listed as "Snapmaker" so I think you will need to go back to them and see if they have updated the plugin.

When I configure any other printer for dual extruders - the tool number is correct regardless of which nozzle is printing the model.
For your printer, when T1 is the primary extruder the gcode doesn't include ANY tool changes and so it is defaulting to T0 because it isn't being told any different.

If everything worked fine in 5.9.0 then I suggest you move back to it for now.  I'll leave this open but it doesn't really look like a Cura problem but rather some incompatibility with either your printer definition files, or with the SnapMaker plugin (both of which come from SnapMaker).

GregValiant, 2025-04-04:
I looked at this some more.
@HellAholic , whenever the model, the support, and the bed adhesion are set to the same extruder, the tool number is not included in the Gcode.  That is not a problem when they are set to T0, but if they are set to T1 the command is missing and the printer starts with T0 because it was never told any different.
With a UM S5 active the problem doesn't seem to occur.  With the other manufacturers I checked - they all experience the problem.
This does not seem to have been an issue in 5.9.
Here is a project file with a Bibo dual extruder.

[20477.zip](https://github.com/user-attachments/files/19602559/20477.zip)

B5TRK, 2025-04-10:
> Thanks for the analysis @GregValiant <https://github.com/GregValiant>  😍
> We've added a ticket to the backlog with the intent to improve this.
> For internal reference CURA-12525 <https://ultimaker.atlassian.net/browse/CURA-12525>
>
> Thanks for the report! 👍
>
>
>
> Thanks for the analysis @GregValiant <https://github.com/GregValiant>  😍
> We've added a ticket to the backlog with the intent to improve this.
> For internal reference CURA-12525 <https://ultimaker.atlassian.net/browse/CURA-12525>
>
> Thanks for the report! 👍

Hi, thank you for the update.

 

Looking at the Snapmaker forum, quite a few users are experiencing this bug in the latest 5.10 release. All have reverted to 5.9. They are concerned that it will be present in future releases. 

 

Thank you in advance

 

Many thanks and Kind Regards

 

 

Barrie Stirk

GregValiant, 2025-04-23:
There is a pull request in for this fix in the Cura Engine.
The workaround fix is to add this to the beginning of your StartUp Gcode

```
{if initial_extruder_nr == 1}
T1 ; Set the extruder
{endif}
```