Ultimaker/Cura #20479: can't load plugins

agrishin, 2025-04-03:
### Cura Version

5.9.1/5.10.0

### Operating System

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

Just installed Ultimake Cura 5.10.0 on Windows 10 (x64) Couldn't get any plugin to load following instructions on the website. Uninstalled and installed 5.9.1 --same issue. Plugins just don't seem to be recognized.

### Actual results

Nothing happens (no Error messages). Just nothing

### Expected results

Expect 'hello world' for simplest example. Expect plugin functionality to show up for other examples

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2025-04-03:
Thanks for the report.
A plugin is a "Cura Extension" and Cura expects that it will be put together correctly.  Extensions typically consist of several files.

Can you load any plugins from the Cura MarketPlace?  The "Printer Settings" plugin is handy to have so see if you can install that one.

agrishin, 2025-04-03:
I can’t even sign in (which I assume you have to do for Cura Marketplace. Please tell me the process if that’s wrong). When I tried to set up an account, it asked for a password (no first-time option to create one). So, I went through the ‘forgot password’ process and never received a follow-up email.

Alex Grishin, PhD
Consulting Engineer
PADT, Inc.
www.padtinc.com
480-813-4884 x133
***@***.***

Simulation – Product Development – Rapid Prototyping

CONFIDENTIALITY NOTICE: This e-mail message and any attachments are for the sole use of the intended recipient(s) and may contain confidential and/or privileged information.  Unless you are the intended recipient, you are hereby notified that copying, forwarding, printing or otherwise disseminating the information contained in or attached to this e-mail is strictly prohibited. If you are not the intended recipient, please notify the sender by telephone, and immediately and permanently delete and destroy all copies and printouts of this e-mail message and/or attachments.

agrishin, 2025-04-03:
Again: Can’t access “Cura Marketplace”, but I can download  ‘plugins’ (wrong terminology?) shown below. Are these all deprecated or something (I don’t see ‘Printer Settings’ there)…?

***@***.***
Alex Grishin, PhD
Consulting Engineer
PADT, Inc.
www.padtinc.com
480-813-4884 x133
***@***.***

Simulation – Product Development – Rapid Prototyping

CONFIDENTIALITY NOTICE: This e-mail message and any attachments are for the sole use of the intended recipient(s) and may contain confidential and/or privileged information.  Unless you are the intended recipient, you are hereby notified that copying, forwarding, printing or otherwise disseminating the information contained in or attached to this e-mail is strictly prohibited. If you are not the intended recipient, please notify the sender by telephone, and immediately and permanently delete and destroy all copies and printouts of this e-mail message and/or attachments.

GregValiant, 2025-04-04:
I've never had an UltiMaker account and I don't think you need one for the MarketPlace when you access it from within Cura (button near the upper right of the workspace).

[image]

Once in the MarketPlace click on the Gear icon.

[image]

agrishin, 2025-04-04:
Following your suggestion to install the "Printer Settings" package from the Marketplace (thanks!  Didn't know I had access to that):
The first image below shows the GUI before adding the package. The second image shows the Marketplace dialog menu after I hit 'Install' on the Printer Settings package. The third image shows the GUI after I restart Cura (no change). It's my understanding that something should have been added to the "sidebar". But I don't see it. I know something got installed because my 'plugins' folder now has a 'PrinterSettingsPlugin' folder. But is something more substantial supposed to happen?

[image]

[image]

[image]

fieldOfView, 2025-04-04:
> It's my understanding that something should have been added to the "sidebar". But I don't see it. 

You have hidden your "sidebar" (also named "printer settings" panel, unless you have the sidebar plugin installed). Click on the bar below the Marketplace button.

fieldOfView, 2025-04-04:
> Couldn't get any plugin to load following instructions on the website. 

I'm curious: what instructions on what website are you following?

agrishin, 2025-04-04:
@fieldOfView: I didn't hide anything, unless you're referring to the pop-out menu shown below. As you say, that's also called 'Printer Settings'. But that looks the same before and after installing the 'Printer Settings' plugin. In other words, I don't see any new functionality. I'm going to let this go because I DO see other plugins working (for example: the 'sidebar' plugin does indeed seem to add a 'sidebar' when I install it from the Marketplace). The instructions I referred to earlier are those found [here](https://github.com/Ultimaker/Cura/wiki/Plugins-And-Packages).

So I'm going to assume adding packages (not sure of the distinction between plugins and packages) from the Marketplace works just fine. What about creating my own plugin (that's my real goal)? How can I go about testing it on my Windows 10 desktop?

[image]

GregValiant, 2025-04-04:
Click on the "Custom" button to bring up the full collection of settings.  "Printer Settings" will be at the bottom of the category list just below "Experimental".  You can click on the "hamburger" icon (setting visibility icon) and select "All" in case some are hidden.

[image]

There are "Plugins" (or "Extensions") that get integrated into Cura.  (Field Of View is the master).
There are also "Post-Processors" that alter the gcode that Cura produces.  They are more of an "occasionally needed" thing and are available under the "Extensions | Post Processing | Modify Gcode" menu and then "Add a Script" in the post processor dialog.

This "Won't load plugins" thing seems to have been sorted out.  I'll go ahead and remove the bug label and close this.  The conversations can continue.

closed 2025-04-04 by GregValiant

agrishin, 2025-04-04:
I agree with that as far as installing extensions from the Marketplace goes. But my original question was more about how to create my own plugin (how load and test plugins created on my Windows desktop). For example, [this extension](https://github.com/Ultimaker/UraniumExampleExtensionPlugin) isn't in the Marketplace list (as far as can tell). Suppose I clone it or download to my desktop and modify it. Now, how do I test and load it?

fieldOfView, 2025-04-05:
> The instructions I referred to earlier are those found [here](https://github.com/Ultimaker/Cura/wiki/Plugins-And-Packages).

The part where it mentions making a zip and renaming that file to `.plugin` is for distributing your plugin. You are supposed to drop the .plugin file onto the buildplate in Cura, as if you were opening a 3d model by drag-and-drop. 

Honestly though, I don't know if the `.plugin` extension even works anymore; that part of the instruction may be outdated. Nowadays plugins are distributed as `.curapackage`, and their format is a bit more involved than the previous `.plugin`, `.umplugin` or `.curaplugin` files.

Here is a bit more to read from a former community plugin developer:
https://thopiekar.eu/cura/plugins/installation/

You do not need to make a `.curapackage` file during development, or to test/use the plugin yourself.

fieldOfView, 2025-04-05:
> Now, how do I test and load it?

Put the files you downloaded in a subfolder inside `plugins` folder inside the Configuration Folder (eg `%APPDATA%\cura\5.9`), where you previously put the `.plugin` file.

agrishin, 2025-04-05:
thanks for replying. I discovered the trick of renaming the compressed files with a .curapackage extension.  However, that doesn’t work either. I DO get the message below, but then nothing happens when I restart. This workflow seems to be broken, as the ‘packages.json’ file never gets updated (in the <configuration folder>). It doesn’t seem to know where to look for the package. Even If I place it it the <configuration folder>\plugins folder
***@***.***
Alex Grishin, PhD
Consulting Engineer
PADT, Inc.
www.padtinc.com
480-813-4884 x133
***@***.***

Simulation – Product Development – Rapid Prototyping

CONFIDENTIALITY NOTICE: This e-mail message and any attachments are for the sole use of the intended recipient(s) and may contain confidential and/or privileged information.  Unless you are the intended recipient, you are hereby notified that copying, forwarding, printing or otherwise disseminating the information contained in or attached to this e-mail is strictly prohibited. If you are not the intended recipient, please notify the sender by telephone, and immediately and permanently delete and destroy all copies and printouts of this e-mail message and/or attachments.

https://thopiekar.eu/cura/plugins/installation/

https://thopiekar.eu/cura/plugins/installation/

agrishin, 2025-04-05:
What do you mean by “You do not need to make a .curapackage file during development, or to test/use the plugin yourself”
How does that work?

Alex Grishin, PhD
Consulting Engineer
PADT, Inc.
www.padtinc.com
480-813-4884 x133
***@***.***

Simulation – Product Development – Rapid Prototyping

CONFIDENTIALITY NOTICE: This e-mail message and any attachments are for the sole use of the intended recipient(s) and may contain confidential and/or privileged information.  Unless you are the intended recipient, you are hereby notified that copying, forwarding, printing or otherwise disseminating the information contained in or attached to this e-mail is strictly prohibited. If you are not the intended recipient, please notify the sender by telephone, and immediately and permanently delete and destroy all copies and printouts of this e-mail message and/or attachments.

https://thopiekar.eu/cura/plugins/installation/

https://thopiekar.eu/cura/plugins/installation/

GregValiant, 2025-04-05:
I've transposed a couple of my post processors into plugins.  They work fine, but I would not use me as an example.

I copied one of the simpler plugins, started remaining things, changed a few lines, and then added my single post-processor file as the main working file of the plugin.  The settings in particular needed translation so they would work from within Cura.  It just took time more than anything else.  My knowledge of QML is real close to zero, and I'm not very good at Python, but by beating on things, adding logging lines to figure out where things were hanging up, and opening and closing Cura 5000 times, I got them to work.
But the plugins I created remain post-processors at their heart.  It's all I needed from them.

agrishin, 2025-04-05:
Understood. Ultimately, I will need to write a plugin that performs some model queries –meaning it will need to interact with the Cura API. But one thing at a time. Right now, I’d be thrilled if I could just write a simple script that does something innocuous (like print ‘hello world’) and have Cura load it. What I’m slowly discovering is that, although there does seem to a pipeline for this, I think its broken.
For example, If I create a simple plugin according to format described here<https://github.com/Ultimaker/Uranium/wiki/Creating-plugins> (or copy one from the github plugins repository<https://github.com/Ultimaker/Cura/tree/main/plugins>), I can’t figure out how to load it ‘manually’. I am aware of the trick of compressing the directory containing the plugin files and renaming the extension (to .curapackage), but that doesn’t seem to work (nothing happens after I drag-and-drop that into the app and restart).

Alex Grishin, PhD
Consulting Engineer
PADT, Inc.
www.padtinc.com
480-813-4884 x133
***@***.***

Simulation – Product Development – Rapid Prototyping

CONFIDENTIALITY NOTICE: This e-mail message and any attachments are for the sole use of the intended recipient(s) and may contain confidential and/or privileged information.  Unless you are the intended recipient, you are hereby notified that copying, forwarding, printing or otherwise disseminating the information contained in or attached to this e-mail is strictly prohibited. If you are not the intended recipient, please notify the sender by telephone, and immediately and permanently delete and destroy all copies and printouts of this e-mail message and/or attachments.

fieldOfView, 2025-04-07:
Just so you know, by replying to the email messages that github sends out when someone makes a new comment to this issue, you are posting to a public website. So your Confidentiality Notice serves no purpose but to obscure your questions.