Ultimaker/Cura #20485: Top/Bottom Line Directions rotating from layer 1

sargonphin, 2025-04-05:
### Cura Version

5,10,0

### Operating System

W11

### Printer

Custom Tronxy X5SA-330 Pro

### Reproduction steps

1. Set more than 2 line directions for skin
2. Slice a model with a large enough skin surface
3. Observe discrepancies in the skin orientation between model and settings

### Actual results

This is not really a "bug", as in, it's not a breaking thing or whatever, but I have noticed that when there are multiple skin line directions, Cura starts counting directly from layer 1 and "rotates" the pattern in the background for each layer, regardless if there is skin or not. This means that the exact pattern is not necessarily respected when reaching the top of the model

[image]

While this is not necessarily detrimental for printing, I would like the top skin layers to start and finish with diagonals in order to fill the entire surface and leave no gap. When parallel directions, if the skin width is not an exact multiple of the layer width, there will be a gap on the entire length

[image]

### Expected results

Ideally, the line direction order would have to be respected regardless of layer height and even more ideally so, resets to beginning when skin surface increases substantially from one layer to the other

### Add your .zip and screenshots here ⬇️

[sandpaper_block.zip](https://github.com/user-attachments/files/19614932/sandpaper_block.zip)
I think in this ZIP file, the line directions aren't the ones I used before. You can just play around with them. Right now I have [45, 180, 90,135]

GregValiant, 2025-04-05:
Thanks for the report.
When I try to open that project file it causes Cura to crash hard.  I imported the model instead.

If you set the "Top Surface Skin Layers" to 1 and then set the "Top Surface Skin Line Directions" to [45] you can dictate the line direction of the top most layer of any skin.

This is one of my demo models and it happens to have "top skins" at different Z-heights.  With malice aforethought, I set the line directions to [45, 180, 90] so that each "top skin" would have lines in different directions.
[image]

Setting the Top Skin layers = 1 and the top line directions to [45] results in each top level being the same direction.
[image]

The "bottom skin" line directions could still be wonky, but the tops should be the same directions.

Is this the sort of thing you are talking about?

sargonphin, 2025-04-05:
It's sort of what I was experiencing, and your fix is something I have considered, except there is a small problem. There is a chance that the layer just below the top skin ends up in the same orientation as the layer below, which would cause horrible surface quality 😁 Here is a really good example of the sort of stuff I'm trying to avoid:

[image]

You can see the infill below. This would have structural strength and... well, completeness problems

[image]

Sorry it's hard to see in this picture, but the lines perpendicular to the wall at the bottom of the picture "overlap" properly, but the lines that are parallel have a gap because the line width is not a perfect multiple of the object width

GregValiant, 2025-04-05:
This is FDM printing, not injection molding.  There are going to be some sort of artifacts present.  Even molded products will have mold parting lines, vent scars and sprue scars.  The companies that do injection molding spend a lot of time on bench work cleaning up their products.
Remember too that all the slicers are written for "all models".  They can't just be good at one particular facet or one design, they are necessarily generic.  In your case you can spend the time moving the [line direction] numbers around to avoid having consecutive layers that run the same way.  It will be somewhat time intensive.  You can also change the layer height by a little bit (like from .20 to .22) and get different result.  That's what I had to do to get the top surfaces all different in that image I posted.

sargonphin, 2025-04-06:
Thanks for the reply, but I'm not certain I understand the point (and the comparison with injection molding). The issue I am having here is common to all models that I import. But that's ok, I suppose I can work with the top surface skin layer settings and figure something out. Too bad that this requires user input instead of being automatic

I'm closing the ticket, if Cura team wants to further investigate, don't hesitate to re-open a ticket and I will provide any information necessary

closed 2025-04-06 by sargonphin

PartySausage, 2025-04-06:
I raised the same issue back in December under https://github.com/Ultimaker/Cura/issues/20000#issue-2721389748. 
I couldn't find a workaround that I was happy with that would give me the results I was looking for on aesthetic prints.
I'm hopeful that one day the Cura dev team will be able to assign some time to take a look at it & make the top layer line directions consistent independent of the layer where the surface is

sargonphin, 2025-04-06:
I tried also the top surface skin layer but it follows the same rotation from layer 1, doesn't work either (same outcome). So now I ended up reverting back to original and going with what Cura can offer which completely prevents using more than 2 angles but, well, that was better than the risk of having 2 layers in the same direction