Ultimaker/Cura #20486: Build from source fails on Windows 10

agrishin, 2025-04-05:
### Cura Version

5.10.0

### Operating System

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

Followed the directions found [here](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source). Step 3 fails with the error listed below.
OS=Windows 10
Cmake Version = 4.0'
Python Version = 3.12
MSVC Version = 17 (2022)

[image]

### Actual results

`FAILED: CMakeFiles/Arcus.dir/src/Socket.cpp.obj
C:\PROGRA~1\MICROS~4\2022\COMMUN~1\VC\Tools\MSVC\1443~1.348\bin\Hostx64\x64\cl.exe  /nologo /TP -DArcus_EXPORTS -D_WIN32_WINNT=0x0600 -IC:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\include -IC:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src -external:IC:\Users\alex.grishin\.conan2\p\proto05d093ec480af\p\include -external:W0 /DWIN32 /D_WINDOWS /W3 /GR /EHsc /O2 /Ob2 /DNDEBUG -std:c++17 -MD /W4 /w14242 /w14254 /w14263 /w14265 /w14287 /we4289 /w14296 /w14311 /w14545 /w14546 /w14547 /w14549 /w14555 /w14619 /w14640 /w14826 /w14905 /w14906 /w14928 /permissive- /showIncludes /FoCMakeFiles\Arcus.dir\src\Socket.cpp.obj /FdCMakeFiles\Arcus.dir\ /FS -c C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket.cpp
cl : Command line warning D9025 : overriding '/W3' with '/W4'
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(110): error C3083: 'system_clock': the symbol to the left of a '::' must be a type
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(110): error C2955: 'std::chrono::time_point': use of class template requires template argument list
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.43.34808\include\__msvc_chrono.hpp(200): note: see declaration of 'std::chrono::time_point'
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(67): error C2512: 'std::chrono::time_point': no appropriate default constructor available
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.43.34808\include\__msvc_chrono.hpp(200): note: see declaration of 'std::chrono::time_point'
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(348): warning C4267: 'initializing': conversion from 'size_t' to 'uint32_t', possible loss of data
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(348): warning C4267: 'initializing': conversion from 'size_t' to 'const uint32_t', possible loss of data
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(560): error C3083: 'system_clock': the symbol to the left of a '::' must be a type
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(560): error C2039: 'now': is not a member of 'std::chrono'
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.43.34808\include\__msvc_chrono.hpp(286): note: see declaration of 'std::chrono'
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(560): error C3536: 'now': cannot be used before it is initialized
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(560): error C2064: term does not evaluate to a function taking 0 arguments
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(561): error C2677: binary '-': no global operator found which takes type 'std::chrono::time_point' (or there is no acceptable conversion)
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.43.34808\include\__msvc_chrono.hpp(296): note: could be 'common_type<std::chrono::duration<_Rep,_Period>,std::chrono::duration<_Rep2,_Period2>>::type std::chrono::operator -(const std::chrono::duration<_Rep,_Period> &,const std::chrono::duration<_Rep2,_Period2> &) noexcept(<expr>)'
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(561): note: 'common_type<std::chrono::duration<_Rep,_Period>,std::chrono::duration<_Rep2,_Period2>>::type std::chrono::operator -(const std::chrono::duration<_Rep,_Period> &,const std::chrono::duration<_Rep2,_Period2> &) noexcept(<expr>)': could not deduce template argument for 'const std::chrono::duration<_Rep,_Period> &' from 'int'
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.43.34808\include\__msvc_chrono.hpp(545): note: or       'std::chrono::time_point<_Clock,common_type<_Duration,std::chrono::duration<_Rep2,_Period2>>::type> std::chrono::operator -(const std::chrono::time_point<_Clock,_Duration> &,const std::chrono::duration<_Rep2,_Period2> &) noexcept(<expr>)'
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(561): note: 'std::chrono::time_point<_Clock,common_type<_Duration,std::chrono::duration<_Rep2,_Period2>>::type> std::chrono::operator -(const std::chrono::time_point<_Clock,_Duration> &,const std::chrono::duration<_Rep2,_Period2> &) noexcept(<expr>)': could not deduce template argument for 'const std::chrono::time_point<_Clock,_Duration> &' from 'int'
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.43.34808\include\__msvc_chrono.hpp(553): note: or       'common_type<_Duration1,_Duration2>::type std::chrono::operator -(const std::chrono::time_point<_Clock,_Duration> &,const std::chrono::time_point<_Clock,_Duration2> &) noexcept(<expr>)'
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(561): note: 'common_type<_Duration1,_Duration2>::type std::chrono::operator -(const std::chrono::time_point<_Clock,_Duration> &,const std::chrono::time_point<_Clock,_Duration2> &) noexcept(<expr>)': could not deduce template argument for 'const std::chrono::time_point<_Clock,_Duration> &' from 'int'
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(561): error C2672: 'std::chrono::duration_cast': no matching overloaded function found
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.43.34808\include\__msvc_chrono.hpp(87): note: could be '_To std::chrono::duration_cast(const std::chrono::duration<_Rep,_Period> &) noexcept(<expr>)'
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(563): error C3536: 'diff': cannot be used before it is initialized
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\Socket_p.h(571): error C2582: 'operator =' function is unavailable in 'std::chrono::time_point'
[5/6] Building CXX object CMakeFiles\Arcus.dir\src\MessageTypeStore.cpp.obj
cl : Command line warning D9025 : overriding '/W3' with '/W4'
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\MessageTypeStore.cpp(25): warning C4267: 'initializing': conversion from 'size_t' to 'uint32_t', possible loss of data
C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\src\MessageTypeStore.cpp(162): warning C4242: 'argument': conversion from 'int' to 'const _Elem', possible loss of data
        with
        [
            _Elem=char
        ]
ninja: build stopped: subcommand failed.

arcus/5.10.0: ERROR:
Package '60f76128b5eaf81f4f1f884f801c871770def54e' build failed
arcus/5.10.0: WARN: Build folder C:\Users\alex.grishin\.conan2\p\b\arcusbc4998b75b3a4\b\build\Release
ERROR: arcus/5.10.0: Error in build() method, line 137
        cmake.build()
        ConanException: Error 1 while executing`

### Expected results

Expected a clean Python build environement

### Add your .zip and screenshots here ⬇️

[image]

keyboard2005, 2025-04-09:
Has the issue been resolved? Compiling it is as difficult as climbing a mountain. I successfully compiled it on Ubuntu 24, but it doesn't work on Windows.

agrishin, 2025-04-09:
No. The issue hasn't been resolved. I've resigned myself to believing no-one has ever tested this on Windows (Linux developers only). I even tried to 'trick' it by attempting to build from source on [MSYS2](https://www.msys2.org/). Even that fails.
Thanks for replying...

keyboard2005, 2025-04-10:
> 不。问题尚未解决。我只好相信没有人在 Windows 上测试过这个（仅限 Linux 开发人员）。我什至试图通过尝试在 [MSYS2](https://www.msys2.org/) 上从源代码构建来“欺骗”它。即使这样也失败了。谢谢你的回复...

I successfully compiled it. The key was to use the x64 Native Tools Command Prompt for VS 2022.
In this terminal environment, I was able to compile successfully. If it still doesn't work, try installing Visual Studio 2022 (version 17.9) using the Visual Studio Installer.
I’m quite confident this was how I resolved the issue.

However, the bad news is that even after successful compilation, you may encounter another problem where the PyQt6 network module (net) cannot be found. I’ve seen this issue mentioned in discussions, but no solution has been provided yet.

keyboard2005, 2025-04-10:
conanfile.py (cura/5.11.0-alpha.0): Retrieving license for wheel
conanfile.py (cura/5.11.0-alpha.0): Retrieving license for zeroconf
conanfile.py (cura/5.11.0-alpha.0): Retrieving license for zipp
conanfile.py (cura/5.11.0-alpha.0): Retrieving license for zope.interface
conanfile.py (cura/5.11.0-alpha.0): Generating aggregated env files
conanfile.py (cura/5.11.0-alpha.0): Generated aggregated env files: ['conanrun.bat', 'conanbuild.bat']
Install finished successfully
(cura_venv) PS C:\Users\samur\Desktop\cura_build\Cura> .\build\generators\cura_venv\Scripts\Activate.ps1
(cura_venv) PS C:\Users\samur\Desktop\cura_build\Cura> python .\cura_app.py
Traceback (most recent call last):
  File "C:\Users\samur\Desktop\cura_build\Cura\cura_app.py", line 27, in <module>
    from PyQt6.QtNetwork import QSslConfiguration, QSslSocket
ImportError: DLL load failed while importing QtNetwork: 找不到指定的模块。
(cura_venv) PS C:\Users\samur\Desktop\cura_build\Cura>

agrishin, 2025-04-10:
with Visual Studio Community 2022 version 17.13.2, my problem still persists (see the output below from the x64 Native Tools Command Prompt for VS 2022). Its the same issue (can't build Arcus).
Do you know if its possible to read an STL file, store it in a MeshData object and query its properties without 'building from source'? In other words, can I do this purely within the Uranium framework without having to connect to a running Cura application (I presume that's what 'Arcus' is for)? If so, I don't think I even need the development environment (I just need to write a plugin, but I'll need a way to test the plugin for specific meshes).

[image]

keyboard2005, 2025-04-11:
> Visual Studio Community 2022 版本 17.13.2

bro try use  Visual Studio Community 2022 v17.9

keyboard2005, 2025-04-11:
> 使用 Visual Studio Community 2022 版本 17.13.2 后，我的问题仍然存在（请参见下面 VS 2022 x64 原生工具命令提示符的输出）。问题依旧（无法构建 Arcus）。 您知道是否可以读取 STL 文件，将其存储在 MeshData 对象中，并查询其属性，而无需“从源代码构建”吗？换句话说，我是否可以纯粹在 Uranium 框架内执行此操作，而无需连接到正在运行的 Cura 应用程序（我猜想“Arcus”就是为此而设计的）？如果可以，我想我甚至不需要开发环境（我只需要编写一个插件，但我需要一种方法来针对特定网格测试该插件）。
> 
> [image]

It sounds incredible, but this will most likely solve your problem.

agrishin, 2025-04-11:
I don't think that's available any more is it? Don't you have to migrate to professional to get versions that old? In any case, I can't seem to do it --either directly from [the downloads site](https://visualstudio.microsoft.com/thank-you-downloading-visual-studio/?sku=Community&channel=Release&version=VS2022&source=VSLandingPage&passive=false&cid=2030), or via the desktop installer

LowPower-Center, 2025-04-12:
You can find v17.9 on this website [downloads site](https://learn.microsoft.com/en-us/visualstudio/releases/2022/release-history#updating-your-installation-to-a-specific-release). 
 It really solves this problem.

agrishin, 2025-04-12:
I'll take your word for it -but those are all the professional version (consistent with my previous comment. You have to pay for a subscription to activate these, right?)

agrishin, 2025-04-12:
Sorry. Scratch my previous comment. Downloading version 17.9 Build Tools (screenshot below) is enough.
Success at last (thank you @LowPower-Center and @keyboard2005)!

[image]

[image]

keyboard2005, 2025-04-12:
> 抱歉。删掉我之前的评论。下载 17.9 版 Build Tools（如下图）就够了。 终于成功了（谢谢）[@LowPower-Center](https://github.com/LowPower-Center)和[@keyboard2005](https://github.com/keyboard2005))！
> 
> [image]
> 
> [image]

Even if it succeeds, there will be other problems. Can you run it normally?

LowPower-Center, 2025-04-13:
#20084 ,this issue  provide a method to solve your problem @keyboard2005 , it works for me.
First, your create a new virtual python env , manually install python packages from two requirements files.
Then,  copy dlls from venv/Lib/site-packages(contains arcus.dll ) to your new python site-packages folder.
You should set environment variable.
Add PYTHONPATH  for  Uranium and Dlls and your new python site-package, like 
```
D:\Github\Cura\Cura-5.10.0\venv\Lib\site-packages;
C:\Users\chen\.conan2\p\uranie414bfe6774dc\p\plugins;
C:\Users\chen\.conan2\p\uranie414bfe6774dc\p\site-packages;
```
After that ,I can run cura_app in Pycharm, but I found another  CuraEngine error.
Last, I compiled CuraEngine 5.10 using VS 2022 v17.9 and finished slicing.

keyboard2005, 2025-04-13:
> [#20084](https://github.com/Ultimaker/Cura/issues/20084)，这个问题提供了一种解决你问题的方法[@keyboard2005](https://github.com/keyboard2005)对我来说，它确实有效。 首先，创建一个新的虚拟 Python 环境，并从两个依赖文件中手动安装 Python 包。 然后，将 venv/Lib/site-packages 目录下的 dll 文件（包含 arcus.dll ）复制到新的 Python site-packages 文件夹。 你应该设置环境变量。 为 Uranium 和 dll 文件以及新的 Python site-package 添加 PYTHONPATH，例如
> 
> ```
> D:\Github\Cura\Cura-5.10.0\venv\Lib\site-packages;
> C:\Users\chen\.conan2\p\uranie414bfe6774dc\p\plugins;
> C:\Users\chen\.conan2\p\uranie414bfe6774dc\p\site-packages;
> ```
> 
> 之后，我在 Pycharm 中可以运行 cura_app，但发现 CuraEngine 又出现了一个错误。 最后，我使用 VS 2022 v17.9 编译了 CuraEngine 5.10，并完成了切片。

是这样的， 我遇到的问题不是这个
File "C:\Users\samur\Desktop\cura_build\Cura\cura_app.py", line 27, in
from PyQt6.QtNetwork import QSslConfiguration, QSslSocket
ImportError: DLL load failed while importing QtNetwork: 找不到指定的模块。
(cura_venv) PS C:\Users\samur\Desktop\cura_build\Cura>
就算编译成功了， 会出现这个报错

keyboard2005, 2025-04-14:
> [@keyboard2005](https://github.com/keyboard2005)：是的，按照所有建议操作后，我仍然遇到同样的错误……

haha This problem seems difficult to solve

subramanya-a, 2025-08-02:
Hello @agrishin, @keyboard2005 
I'm observing same error in Win10, any solution/fix available to resolve this issue ?