Ultimaker/Cura #20487: ultamaker cura has stopped working and is seriously pissing me off

RazorFew, 2025-04-05:
### Cura Version

5.10.0

### Operating System

WINDOWS 10

### Printer

Anycubic Vyper 

### Name abnormal settings

Not that im aware of i have sliced and printed this file before 

### Describe your model

its a model i built to hold a tuner on the bars of my harley 

### Add your .zip here ⬇️

[Another Fucking PowerVision v7.zip](https://github.com/user-attachments/files/19617951/Another.Fucking.PowerVision.v7.zip)

GregValiant, 2025-04-06:
Thanks for the report.
I didn't find any errors in the model and it sliced fine for me with 5.10 and on my Win10 Pro system.

[image]

Set Cura up to slice and then force the failed slice.  Then use the "Help | Export file for Technical Support" command.  Zip that 3mf file and post it.  That will contain your printer, your settings, and your log file.  Maybe the log file will have a clue in it.

closed 2025-04-20 by github-actions[bot]

a4jp-com, 2025-04-20:
If you select the body then move tool you should be able to place the body on the bed.