Ultimaker/Cura #20491: Copy/paste no longer works

baboo1969, 2025-04-06:
### Cura Version

5.9.1

### Operating System

Windows 11

### Printer

Elegoo Neptune 4 Max

### Reproduction steps

Ctrl-c and Ctrl-V, or contextuals copy-paste no longer works!!! Did you made some tests?

### Actual results

Do not work

### Expected results

copy paste

### Add your .zip and screenshots here ⬇️

none

GregValiant, 2025-04-06:
Thanks for the report.
I have no idea what could cause that.  I've never had the problem and I just checked and it isn't occurring on my Win10 laptop.  It could be a Windows thing specific to your computer.
I don't know that your log would show anything but you could zip it and post it ("Help | Show Configuration Folder" and "Cura.log") and the Cura team will take a look at some point.

baboo1969, 2025-04-07:
Hi,

  i have a brand spanking new Win11 installation!

Next to my message, i found that the software had paste the object...  the pasted objects are far up in the air!

I'm using an Elegoo Neptune 4 max, and it paste objects 2 to 3x the height of the printer in the air (z at more than 1000mm).  Don't tell me it's my Windows!

GregValiant, 2025-04-07:
Now see, you didn't explain that in the first post, you just said _"copy-paste no longer works!!"_.  But it does work, it just doesn't work correctly.
Someone from the Cura team will take a look.

baboo1969, 2025-04-07:
Hi,

  because i saw it after opening the ticket