Ultimaker/Cura #20494: Intercambio de cabezales en 5.10

asantoroar, 2025-04-07:
### Cura Version

5.10

### Operating System

W11

### Printer

Hellbot Hidra Plus

### Reproduction steps

Segmente utilizando el cabezal 2 de la impresora y deshabilitando el cabezal 1.
Al imprimir, lo hace en el cabezal 1.
Volvi a generar la misma impresion en Cura 5.8 y funciono correctamente 

### Actual results

c

### Expected results

Que imprima en el cabezal correcto

### Add your .zip and screenshots here ⬇️

d

GregValiant, 2025-04-07:
Thanks for the report.
This got broken in 5.10.0.  It worked good in 5.9.0 and 5.9.1 so you an revert to one of those as a workaround.
This is a duplicate of #20484 and #20477 so I will go ahead and close this one.  Thanks for reporting even if you weren't the first.

The Cura team will take a look.

closed 2025-04-07 by GregValiant

GregValiant, 2025-04-23:
There is a pull request in for this fix in the Cura Engine.
The workaround fix is to add this to the beginning of your StartUp Gcode

```
{if initial_extruder_nr == 1}
T1 ; Set the extruder
{endif}
```