Ultimaker/Cura #20498: Filament stuck after ironing

MDA-13, 2025-04-09:
### Is your feature request related to a problem?

I couldn't find such a topic (if I did not spot it, I apologize), so I'm opening a new one.

Unfortunately, it has happened several times that during the ironing operation of an intermediate level of the model (when the filament feed was practically zero), the printing filament overheated and deformed in the print head, which clogged the filament path, making it impossible to continue printing, as the filament was stuck.
However, I can't find a setting option in Cura where the head temperature during ironing could be adjusted.

How can I solve this problem without the head temperature being adjustable?

Thanks in advance,
Attila

### Describe the solution you'd like

E.g. could this simple parameter be made adjustable in the next update?

Or

I understand that with a reduced head temperature, the ironing efficiency also decreases, since it is more difficult to melt the already printed material with a cooler head. But e.g. by completely pulling the strand out of the heated part of the head while ironing, the problem may be remedied without reducing the head temperature, which I think would be a good solution even at the cost of having to wait a few seconds for the head to reheat before continuing.
What do you think about this?

### Describe alternatives you've considered

I will not use ironing function in the future but this is not a real alternative...

### Affected users and/or printers

Anyone who wants to use ironing function.

### Additional information & file uploads

My system:
OS: Windows 10/64
HW: Ultimaker 3
SW: Ultimaker Cura v5.5.0

GregValiant, 2025-04-09:
I know this problem and there are a couple of different variations as to why it occurs.

- Too many retractions in a small area.
- Ironing with very low flow
- Hot end fans are obstructed, dirty, or inoperative.

There is nothing in a normal Gcode to differentiate "ironing" from any other "skin" area.
There is a setting for "Ironing Jerk" that can be used so something gets inserted when ironing occurs.  Then you can use "Search And Replace" to add M104 lines.  It is a workaround but it is something you can try today.

For this example I have "Print Jerk" set to 8 and "Ironing Jerk" set to 7.9 (just to make it obvious in the Gcode).
Lets say my printing temperature is 210 and I want to iron at 190.  I would use two instances of Search and Replace.

Using one instance of "Search and Replace" like this:
Search:  M205 X7.9 Y7.9
Replace:  M205 X7.9 Y7.9\nM104 S190
Use Regex:  False

And a second instance:
Search:  M205 X8 Y8
Replace:  M205 X8 Y8\nM104 S210
Use Regex:  False

Results in something like this:
M205 X7.9 Y7.9 >>> jerk change for ironing
**M104 S195**   >>>>> Added temperature change for ironing
...
Ironing moves
...
M205 X8 Y8 >>> normal print jerk
**M104 S205** >>>> Added reset to normal print temperature

sargonphin, 2025-04-09:
If I could give a piece of advice, I had the same fear when I enabled ironing for my printer, so what I did is that I set one top surface skin layer and I set the flow to 93%. Then I set the ironing flow to 15%, which keeps the filament flowing "fast enough" to prevent melting everywhere. Gives really good results! Here are the settings I use, feel free to experiment around it:

[image]
[image]

And here is the results I get from this:

[image]

With black filament, you get a very carbon-fiberey look to it and it's stunning