Ultimaker/Cura #20499: Bizarre ghost lines when using ultra-fine layers

sargonphin, 2025-04-09:
### Cura Version

5.10

### Operating System

W11

### Printer

Modified Tronxy X5SA blablabla

### Reproduction steps

1. Set layer thickness to a very low value, eg. 0.08mm
2. Observe some bizarre ghost lines that try to reach the origin coordinates (except on Z)
3.  Move around/rotate the model and observe that the lines always point towards origin

### Actual results

Something really strange appears when setting the layer height to 0.08mm and slicing some models

[image]

These lines appear and seem to be pointing towards printer origin, except on the Z axis. Moving around the object doesn't remove the lines, and they also point to the origin

[image]

I'm unsure if this is just model geometry, so far it appears only with Benchy but I will test some more models in a bit. This is completely gone when setting layer height to something higher. I thought that you guys might want to have a look just for debugging purposes

### Expected results

Well, obviously, no ghost lines 😁 But this is not a big problem for me as I very rarely print at such a layer height

### Add your .zip and screenshots here ⬇️

[3DBenchy.zip](https://github.com/user-attachments/files/19671753/3DBenchy.zip)

PS: @GregValiant mighty impressive patience you've got there... I always loose my mind when I read half of the tickets in there 🤣

GregValiant, 2025-04-09:
I'm retired.  It gives me something to do, and I occasionally learn something as well.

In this case, I'm not learning anything.  First off that Benchy has errors and they should really be fixed.
Next is the weird lines.
Yes, they do appear in the Gcode.  That's a bad thing.  This does seem to be a Cura bug and not related to your video sub-system.

- I can make those bad lines vary by changing the "Minimum Bridge Wall Length".
- If the "Bridge Settings" are turned off, the lines go away.
- If I switch to the regular "Extra Fine" profile (.06 layer height) and turn the Bridge Settings on - those lines are not present.

I've seen this a few times in the last month or so and it appears that something might be corrupt within your "1 - Accurate" custom profile and especially in the Bridge Settings.

For the image below I fooled with the settings a bit.  You can see that we have an extrusion from the model to Z0 Y0 and a travel returning to the model.  I can tell from the width and height that these are bridge lines.
[image]

_**Did you happen to import that profile from a previous version of Cura?**_  Maybe something didn't come in correctly.  That could happen if there were variable names changed from one version to the next and they weren't handled correctly, or an error should have been trapped but wasn't.

For now, I would just delete the "1 - Accurate" profile and rebuild it.  That should keep the problem from reoccurring on your machine.

The Cura team will take a look.

sargonphin, 2025-04-09:
I made profile earlier this afternoon. In fact, it's the first model I sliced with this profile, it's not imported from previous versions. Tomorrow i will try to analyze the profiles and see if i can fix that manually

GregValiant, 2025-04-11:
This would need to be addressed  within the overrides section of your printer definition file.
As you modify a printer past it's stock configuration, you really want a custom definition file.  It can be based on your stock definition, but the "overrides" sections is going to grow.  I've added about 15 lines to my overrides so that Cura sets up the way I want it.  For example I have "group outer walls" turned off by default and I have "skirt_height" set to "1" (because I don't bite my fingernails).
I have also altered the stock StartUp and Ending Gcodes because I simply don't use the Creality defaults.  When I update Cura, I get my settings and not Creality's.

In this case you would want to add a new override:
"coasting_volume": { "default_value", "0.001" },
and maybe
"retraction_amount": { "default_value", "0.80" }

When altering a definition file you must be very careful about punctuation.  Any added line must end with a comma EXCEPT THE LAST ONE which must not end with a comma.  Just pay attention.  Refer to the settings in "fdmprinter.def.json" for the setting names and properties but make all changes as overrides in your custom definition file.  Never change "fdmprinter" itself or you will find out why I always say "Everything affects Everything".
After altering a file (renaming is surprisingly fussy and can take a couple of tries) put it into the "definitions" sub-folder of your configuration folder.

sargonphin, 2025-04-11:
What would this address exactly? The profile for my printer is entirely made from a blank slate. Start and end GCODE are fully customized, there is pretty much nothing original left but the aluminum frame of this poor Tronxy 😄 I never needed to alter the definition file, so far this bug is pretty mundane. I just set the coasting to 0,001 which practically does not impact anything on the print

And this is the only printer I use with Cura, so editing the defaults is not useful in our case

GregValiant, 2025-04-11:
When you update Cura is your machine definition file reverting to the original TronXY version, or are your customizations remaining in place?

Sometimes I have a skewed view because I open so many 3MF files created by other people that I can over-write my own printer.  At that point I have to install another instance of "creality_greg3pro" to get all my stuff back the way I like it.

sargonphin, 2025-04-11:
The machine definition stays (so far at least). Also I would suggest installing some small virtual machines using Virtual Box, use a minimalistic image of Windows 10 and 11 (you may have to do that yourself) and use it to test cross-platforms and cross-versions ;)

Worst case, I export my printer profiles and start/end GCODEs so I can re-create everything quickly, should something go south

GregValiant, 2026-09-08:
I'll go ahead and close this.

closed 2026-09-08 by GregValiant