Ultimaker/Cura #20503: Cure Version

Pete209a, 2025-04-11:
### Cura Version

5.10

### Operating System

Windows 10

### Printer

Creality Ender-3 V3 KE 3D Printer

### Reproduction steps

No big issue just i was asked to upgrade to version 5.10 yesterday which i did twice but when i start Cura it shows 5.91?

Not sure if it did ever upgrade, i didnt un-install as i want to keep my settings

Best Regards

Peter

### Actual results

As above

### Expected results

Expected to see 5.10 at startup

### Add your .zip and screenshots here ⬇️

Nothing more to add

GregValiant, 2025-04-11:
Thanks for the report.
Did you use the official download from here at GitHub?  It's [HERE](https://github.com/Ultimaker/Cura/releases) under "Assets".
The installation file you download includes the version number so check your download folder and see if you grabbed the correct one.
I am also running Windows 10 and I didn't have a problem with "UltiMaker- Cura - 5.10.0-win64-X64 **_.EXE_** ".
I have never used the "UltiMaker - Cura - 5.10.0-win64-X64 **_.MSI_** " installer but it would be extremely odd if it was different.

The opening splash screen of Cura should match what you see with the "Help | About" command.

[image]

Pete209a, 2025-04-11:
Hi Greg

Thanks for the Info, i only did the update as Cura pointed me to it as i opened it  - Here attached are some files, i am running a 64 bit version of Windows 10 and tried both the exe and msi files, as my initial all is working fine just strange please see attached screenshots

[image]
[image]
[image]

Update >>> Thank you have updated from the link you gave for github and all fine now showing 5.10 , might be worth noting incase others have issues !!!!

Many thanks

Pete