Ultimaker/Cura #20505: Please add Print Settings to gcode file

alfaniner, 2025-04-11:
### Is your feature request related to a problem?

Can't save a Print Setting Profile as a permanent fixture. If any settings are changed, they are automatically saved. I need to see what the original gcode settings were when I go to reprint an item.

### Describe the solution you'd like

I had perfect settings for an item and filament  -- I saved the gcode and make a Profile. But I've altered the Profile settings for another project and didn't Save them, but quitting and restarting and pulling up the original Profile now has the new settings in it and I forgot what the perfect ones were when I wanted to reprint that first item.
Can you add a set of readable comments at the beginning with all the Print Settings? So I could recover them from the gcode file if needed. Thanks.

### Describe alternatives you've considered

Saving a Profile, restarting and using that Profile, changing settings, quitting (no Save), and restarting brings up the changed settings. I didn't want to Save over the original Profile.

### Affected users and/or printers

Anyone who uses Cura Ultimaker.

GregValiant, 2025-04-11:
This already exists and there is a plugin or two if this is not sufficient.

The settings are stored at the end of any gcode file.  It looks odd because it is "importable" as a "Profile" and so the "SETTING_3" lines are not really intended for a person to read, but for Cura to read.

;SETTING_3 {"global_quality": "[general]\\nversion = 4\\nname = GV PLA 50\\ndefi
;SETTING_3 nition = creality_greg3pro\\n\\n[metadata]\\ntype = quality_changes\\
;SETTING_3 n**quality_type = standard**\\nsetting_version = 25\\n\\n[values]\\nlayer
;SETTING_3 _height = 0.25\\nlayer_height_0 = 0.25\\nmaterial_bed_temperature = =
;SETTING_3 extruderValueFromContainer(0,\\"material_bed_temperature\\",4)\\nretr
;SETTING_3 action_combing = all\\nspeed_slowdown_layers = 0\\nsupport_enable = = True

Looking at the above, I have made "quality_type" bold and you can see that I started with "Standard".
Everything below that are the settings that deviate from whatever the default is in "Standard".

[general]
version = 4
name = GV PLA 50
definition = creality_greg3pro
[metadata]
type = quality_changes
**quality_type = standard**
setting_version = 25
[values]
layer_height = 0.25
layer_height_0 = 0.25
material_bed_temperature = = extruderValueFromContainer(0,\\"material_bed_temperature\\",4)
retraction_combing = all
speed_slowdown_layers = 0
support_enable = True

So go to "Manage Profiles" and "Import" the Gcode file of your choice.  The new profile will have the same name as the Gcode file.  Then you would need to enable it in Cura.

GregValiant, 2025-04-28:
@alfaniner Is that something that would work for you?  If not, could you explain further what exactly you are looking for?

alfaniner, 2025-04-28:
Thanks for being diligent on following this up. I was looking for an easy, English-readable list, just like the one you see in the Settings you can change for a Profile. But I have stabilized my current Project and don't want to mess with Settings to test things out at this time, so I'll consider the matter closed.

closed 2025-04-28 by alfaniner