Ultimaker/Cura #20509: Profile name

lebele79, 2025-04-14:
### Is your feature request related to a problem?

Every time I want to check what profile a specific gcode was generated with, there is a problem.

### Describe the solution you'd like

Is it possible to add a system variable that would be the name of the profile used?

### Describe alternatives you've considered

I noticed that at the end of the .gcode file in settings 3 you can find the profile name, but there is nothing you can do to make this data at the beginning of the file.

### Affected users and/or printers

I think everyone will benefit from this.

GregValiant, 2025-04-14:
There already is one (it isn't easy to find).
If you add this to your startup gcode it will add the custom name:
{quality_changes_name}.

I'll go ahead and close this.
I'm going to add the "Custom Name" to the post-processor "DisplayInfoOnLCD".  It adds a few things to the beginning of the Gode anyway and in the future it will look like this:

;FLAVOR:Marlin
;TIME:9897
;Cura Time Estimate: 2hr 44min 
;Extruder 1 (T0)
;  Filament used: 20.67m
;  Filament Type: PLA
;  Filament Dia.: 1.75mm
;  Nozzle Size  : 0.4mm
;  Print Temp.  : 250°
;Layer height: 0.28
;Initial Layer Height: 0.32
**;Base Quality Name = 'Standard Quality'   Custom Quality Name = 'GV PLA 50'**
;Model List: ['GV_ComboShape.stl']

closed 2025-04-14 by GregValiant

GregValiant, 2025-04-19:
The pull request was approved and the next release of Cura will have this new and improved "DisplayInfoOnLCD" script.  You can download and unzip this and place the ".py" file into your installation folder of Cura and over-write the existing file.
On my Windows machine and Cura 5.10.0 the install folder is:
"C:\Program Files\UltiMaker Cura 5.10.0\share\cura\plugins\PostProcessingPlugin\scripts"

[DisplayInfoOnLCD.zip](https://github.com/user-attachments/files/19821010/DisplayInfoOnLCD.zip)