Ultimaker/Cura #20513: UltiMaker Cura 5.10.0 open multiple stl files

Stoomtreinen, 2025-04-16:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Creality Ender V2

### Reproduction steps

UltiMaker Cura 5.10.0 If I open multiple stl files at the same time via the explorer,
UltiMaker Cura 5.10.0 closes itself without an error message.
If I open the stl files 1 by 1 it goes fine Please solve.

### Actual results

UltiMaker Cura 5.10.0 closes.

### Expected results

Open multiple stl files at the same time via the  windows explorer.

### Add your .zip and screenshots here ⬇️

Can't give screenshot Cura close without error.

GregValiant, 2025-04-16:
Thanks for the report.
This might be specific to your computer.  I just opened 20 STL files at the same time without a problem.
If I right click on 20 files in Explorer and select "Open with Cura 5.10.0" then each file will open in a separate instance of Cura, but there is no crash.
Post your Cura.log file.  Maybe it will have a clue in it.

Stoomtreinen, 2025-04-17:
[cura.log](https://github.com/user-attachments/files/19794291/cura.log)
I open the follow stl files, zip.file

[Wasstraat.zip](https://github.com/user-attachments/files/19794296/Wasstraat.zip)
Computer:
Windows 11 - 32GB Ram en 1000GB Rom

GregValiant, 2025-04-17:
I'm not having any trouble with it.
If I select all the files in Explorer and right-click and select "Open with Cura 5.10.0" then all the files open, but each model is within it's own instance of Cura.
[image]

If I use the "File Open" command in Cura and select all the files then all open (very quickly) in a single instance of Cura.
If I drag all the files into Cura they all open in that single Cura instance.
[image]

I don't see any specific error in the log file either.  If it's a Windows error then it won't be recorded in the Cura log.  There are some references to OpenGL and to QML which is the controls app that Cura uses.
(BTW my computer is a second hand machine with 8mb of RAM running Win 10 Pro.)

Someone from the Cura team will take a look.