Ultimaker/Cura #20515: support bottom distance not working correctly

jurgenvdb, 2025-04-16:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

I am printing with Ultimaker PLA, using a generic PETG as an interfacing layer for the support to be able to remove it easily. To be able to use it to this use-case I changed the support Z distance to 0, so the support is right op to the part i want to print for the cleanest results.

### Actual results

When i sliced the part i didn't notice anything unusual, until the print failed. I looked at the sliced model and noticed a gap between the top of the print and the support, even though I set the support Z distance (and thus the support bottom distance) to '0.0'. when changing this value, the distance between the model and the support didn't change, but the support floor thickness did seem to change (looking at the results of the sliced model, I didn't change this setting myself). hereby I conclude the support bottom distance is somehow not calculated when sliced?

### Expected results

when changing the support bottom distance, I expect the distance between the top of the model and the bottom of the support to be the distance I just set, which is not the case.

### Add your .zip and screenshots here ⬇️

[image]

[Project file](https://github.com/user-attachments/files/19777717/UMS3_support_bottom_distance_bug.zip)

GregValiant, 2025-04-16:
Thanks for the report.
This looks like a bug to me.  This is the project but I have turned on the interfaces and set the support Horizontal Expansion to zero for a better view of what is going on.
[image]

That is with top and bottom gaps set to 0.0 and the interface thickness at 0.4.  Increasing the interface thickness things get even odder:
This is with the top and bottom air gaps at 0.0 and the interfaces at 1.0 thick.

[image]

@Hellaholic or someone else from the Cura team will take a look.
BTW I get the same effect with my Ender active.

HellAholic, 2025-04-19:
@GregValiant This looks like an issue we have open on the backlog (have to sift through to see if I can find it).
So this is mainly a result of the code (normal support implementation that I'm trying to get refactored / rewritten) trying to generate more layers than the actual gap OR the support layers leaking through the model and getting removed and taking the good layers with them OR any number of different combination of situations (just guesstimating without looking in the code).

Side note, if you have cases where normal support is more favored (other than using PVA as support material) let me know.
Current focus as far as I know is to get the tree support polished instead of rewriting the normal support as it offers faster print and uses less material with the same function, but I've been on vacation for a couple of weeks.

In the meantime:
To do a setting/profile adjustment and get around the issue, I would recommend setting the support roof thickness and support floor thickness individually and checking the preview to find the best combination.

GregValiant, 2025-04-19:
I thought it was pretty odd that it happened with one model but not the other.  The "bleed through" thing might be the problem as the affected model has that narrow shelf with support below and above.

(I use "normal" supports a lot more than I use tree supports.)