Ultimaker/Cura #20517: Create a Printer Profile for the AnyCubic Kobra 3 Max

TomLovering, 2025-04-17:
### Is your feature request related to a problem?

There is no Printer Profile for the AnyCubic Kobra 3 Max

### Describe the solution you'd like

Create a Printer Profile for the AnyCubic Kobra 3 Max

### Describe alternatives you've considered

Shooting myself because I am so bummed without this profile

### Affected users and/or printers

Everyone who buys the Kobra 3 Max

### Additional information & file uploads

When this gets done please let me know. Wish I could do this myself but I am totally clueless about this sort of thing. My name and e-dress is  Tom Lovering,  tlovering@mac.com.  Thanks in advance for doing this.

GregValiant, 2025-04-17:
All the definition files come from community members and/or users of the particular printer and/or occasionally from the manufacturer.
If someone were to submit the files then it would certainly be considered for inclusion.

cellocgw, 2025-05-16:
The headache here is that Kobra 3 Max expects a rather lengthy, customized header (not to mention a thumbnail image!) on the gcode files submitted to it.  If I can figure out exactly what in the header AnyCubicSlicerNext generates is *not* specific to the STL being sliced, I'll post it up here.

TomLovering, 2025-05-16:
Thanks Carl. Your efforts are truly appreciated.

Sincerely,
'
’Tom Lovering

cellocgw
 left a comment 
(Ultimaker/Cura#20517)

SamBkamp, 2025-06-15:
sorry to necro this issue but I just got my paws on a kobra 3 v2 and I wanted to see if I could help solve this issue. First it seems to encode an image file (so far i've only encountered pngs) between `; thumbnail begin SxS N` and `; thumbnail end`. Its b64 encoded. I wrote a little python script to decode that into bites and store it on disk, im sure you could very easily reverse that process to inject images into cura-generated gcode files. The repo is [here](https://github.com/SamBkamp/AnycubicSlicerImageIntractor.git). An injection script will be added to the repo when I get around to it. 

After this, there seems to be a header that is pretty consistent across gcode files:
```
TYPE:Custom
G9111 bedTemp=65 extruderTemp=230 ;CUSTOM COMMAND, NOT DEFAULT MARLIN
M117 ;display LCD message
M900 K0.05 ;linear advance factor, ive only seen this set to k0.05

;START HEADER 
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion {NOT INCLUDED WITH CURA KOBRA 2 SETTING, USES ABSOLUTE}
T0 ;NOT INCLUDED WITH CURRENT KOBRA 2 SETTING - (set or report the current extruder or other tool)
M107 ; turn fan off (???) - NOT IN DEFAULT KOBRA 2 SETTING
```

I've added some annotations and compared to the header code that came with my Kobra 2 profile that ships with Cura. 

I also found gcode annotated as wipe which might be useful 
```
;WIPE_START
G1 F14400
G1 X99.553 Y118.278 E-.08276
G1 X99.817 Y118.115 E-.09825
G1 X100.119 Y117.987 E-.10387
G1 X100.598 Y117.878 E-.15556
G1 X101.731 Y117.809 E-.35956
;WIPE_END
```

I am now going to try and change my cura profile to include the header above instead of the Kobra 2 header

cellocgw, 2025-06-15:
I agree with your findings, and thanks indeed for the thumbnail generating code!  
A gentle reminder to other users:  take a look at all the speed settings in Anycubic's "slicerNext" app.  No point in having a Kobra 3 if you run it slow :-) .

Personally, I like the thicker-walled tree supports slicerNext creates vs.  Cura's thin walls.  I don't see a setting in Cura to change the tree-wall thickness. 

Carl
PS: the opposite of "extract" is "insert," but I like the creativity of "intract".

> sorry to necro this issue but I just got my paws on a kobra 3 v2 and I wanted to see if I could help solve this issue. First it seems to encode an image file (so far i've only encountered pngs) [...]

> I am now going to try and change my cura profile to include the header above instead of the Kobra 2 header

cellocgw, 2025-06-15:
Hi Tom,
If you go , in Cura to Preferences / Printers and click on the printer you are using, then on "Machine Settings," you'll see where you can edit the "Start Gcode"  and "End Gcode" preamble and suffix.   

[image]

I think the Kobra 3 will run without the thumbnail image included in the gcode file -- the only difference will be that the file menu will display a generic icon rather than a picture of the item to be printed.   

regards,
Carl

> Sam,
> 		This is Tom Lovering. I am the end user that put in the request for the Kobra 3 profile in Cura. Thank you for your efforts with this, it is hugely appreciated. Very unfortunately for me I know zero about computer code and so your instructions here are all Greek to me. Will you and/or your team let me know when Cura is updated with the Kobra 3 profile so I can start using it.
> 
> Again, thank you. I wish there were some way I could help out with this.
> 
> Sincerely,
> 
> Tom Lovering
>

SamBkamp, 2025-06-16:
@TomLovering I have created an initial config for the Kobra 3 v2 with buildplate and everything, but its still not perfect. Though if youre interested in using it (and potentially helping me with feature requests) you can check it out [here](https://github.com/SamBkamp/kobra3v2_cura_conf)

HellAholic, 2025-06-16:
> trying to compile Cura from source because I don't think you can plop files in a mounted appimage (im on linux).

You can just drop the files in the configuration folder for linux I believe it's `~/.config/cura/<version>`
Each file should be placed under the corresponding folder that matches the file's content:
- Printer definition -> definitions
- Printer extruder -> extruders
- extruder variants -> variants/manufacturer
- ...
and so on.

SamBkamp, 2025-06-20:
just a quick update @TomLovering. I have submitted a PR with my working profile for this printer. I'm not sure if/when this will go into the main release, but in the mean time you can use the files in that Pull Request

https://github.com/Ultimaker/Cura/pull/20703

GregValiant, 2025-12-14:
Any definition file is just a way to "automatically" alter Cura for a particular printer.  Among the settings is the physical size of the printer build volume (in "Manage Printers" and then "Machine Settings").  You can alter them to what you require.

The "Printer Settings" plugin is available in the MarketPlace.  If you install that you will have access to Printer Settings that are generally only set by definition files (Max Feedrates, the Accel and Jerk settings, etc.).

In addition you can add overrides to a definition file and personalize it.  I won't go into how here, but it isn't rocket science.

HayatoTenryuu, 2025-12-14:
> Any definition file is just a way to "automatically" alter Cura for a particular printer. Among the settings is the physical size of the printer build volume (in "Manage Printers" and then "Machine [...]

That's fine and all, I was just wondering if most of those settings are the same for the Kobra 3 and Kobra 3 Max. Judging from pictures, it looks like they have different printheads, so I may need to measure those by hand. I can easily get the build volume specs from the website, but I'm not sure how much is different between the two.