Ultimaker/Cura #20519: Hot end starts and stop on the exterior wall, causing under extrusion holes.

stang1966gt, 2025-04-17:
### Cura Version

5.10

### Operating System

Mac OS Sonoma

### Printer

Cr-10 Klipper

### Reproduction steps

In the Preview tab
The hot end prints without hesitation along the outer wall.
It is a simple 2 exterior walls and 2 interior walls on this STL.

The Z seam is set to the back face, sharp edge.  This is not the issue

I only print the first 8mm of the SLT, with half of it buried below the build plate.  Trying to understand this issue.

Overture PETG Black

### Actual results

When the printer prints, it consistently stops and restarts, similar to a Z seam, in 2 consistent locations of the exterior wall.  As the Z height changes, the stop and start location change, see photo.  This is causing under extrusion voids with each stop and start.  Likewise, some streak marks from filament heat soaking.

### Expected results

The Exterior wall should be printed in one smooth pass, without stop/start and the voids / heat soak streaking.

### Add your .zip and screenshots here ⬇️

[image]
If the green axis is 6 o'clock, the stop/start happens at 6 & 8:30
Under extrusion issues
[image]

In the video zip, you can see the start/stop on the outer wall, but then continue to do a clean pass on the inner wall.  
[Cura start stop.mov.zip](https://github.com/user-attachments/files/19801584/Cura.start.stop.mov.zip)

[image]
This print is what is currently under the bed in the Project file.  I started the print roughly 10mm below the top of this print.  You can see the shiny heat soak filament start and stop points, matching the video.

GregValiant, 2025-04-17:
Thanks for the report.
The video is nice but doesn't help.  We need a project file and (in this case) that gcode file in order to investigate.

I don't like to guess, but in this case I'll guess that the starts and stops are not in the gcode and it is the Klipper interface that is causing the delays.  UltiMaker Cura does not natively support Klipper.

Post the project and gcode files.  Maybe there is something going on.

stang1966gt, 2025-04-18:
Hi Greg,
I couldn't see how to add additional docs to the report, so it is attached in this email.
Thanks,Will

GregValiant, 2025-04-18:
Use the "Help | Export file for Technical Support" command.  The 3mf file will contain a project file and your Cura.log file.
In Windows Explorer right-click on that 3mf file and select "Send To..." | "Compressed (zip) Folder".  You can drag the gcode file into that new zip folder and then post the folder here (directly to the web site as GitHub doesn't allow posting files via e-mail).

FYI Myself and another member on the UM Forum recently put a lot of time into getting a simple command to work with Klipper (it happened to be M84).  He ended up having to write a macro for it.  There may be other commands that are not supported within Klipper or their macros need some tweaking to work with vanilla Marlin (which I assume you have Cura set to).

Sometime stuttering can be a problem if the "Maximum Resolution" is set too low in Cura.  If it's around 0.40 or higher it shouldn't be a problem.

stang1966gt, 2025-04-18:
[Send to Cura for exterior wall streak and stop n start.zip](https://github.com/user-attachments/files/19815102/Send.to.Cura.for.exterior.wall.streak.and.stop.n.start.zip)  Attached is the project file and Gcode for the stop/starts and body streaks.  

I did not see the PASTE note below this box yesterday.  I thought it was a text only box.  Got it now :)

stang1966gt, 2025-04-18:
Max Resolution was set to 0.25.  Below your suggested 0.4.  I will try later this evening.

[image]

stang1966gt, 2025-04-19:
I used the Maximum resolution 0.40
Removed all the Retractions, 3 boxes if I recall correctly
Ran the print again and it still stops in the same locations
I do get stringing and ugly surfaces on the left and right corners.

History:
I inserted a pause, post gcode modifier, since I didn't want the print to keep printing.  Just wanted a few layers and then make changes, since I was originally having a wall adhesion issue (still do) and was trying different Flow%, speeds, widths.  The stop/start was noticed during wall adhesion tests... and then it matched up why my full body prints had heat streaking.  I tried all the different types of Pauses in the Post Gcode, but haven't found the correct one to do so.

I was also playing with 4 retractions a few days ago via post gcode.  Strange, no changes in stringing.

I attached the gcode file without retractions.  I also notice when I run a single wall Zbanding test, the Z seam either gets extra junky or under extruded.

Ideas what else I can test for you?

[CCR10_SuperVee No retract No Post Gcode.gcode.zip](https://github.com/user-attachments/files/19817633/CCR10_SuperVee.No.retract.No.Post.Gcode.gcode.zip)

The good old Ender 3 is hard to beat :)

GregValiant, 2025-04-19:
I think the best thing for you to do is ask for advice on a Klipper help site.  My Ender gives me really good prints running the 5 year old stock 8bit firmware.  I don't know Klipper, but from what I've heard of it, it's more of a framework that requires macros to define the behavior of certain commands in order to "flesh it out".

I'll leave this open for you and the Cura team will take a look.  I don't see a bug here as it seems to be a problem between the Marlin output from Cura, and the way your Klipper install is translating that gcode and sending it on to the printer.

stang1966gt, 2025-04-19:
Thank you.  I did a small 3D geometry, line and two non tangent arcs.  It printed without issue.

[Simple 3D geometry test.3mf.zip](https://github.com/user-attachments/files/19818436/Simple.3D.geometry.test.3mf.zip)

LilBub, 2025-04-23:
Out of curiosity - are you using the Klipper plugin for Cura?

stang1966gt, 2025-04-23:
I have it installed but do not use it to input values to klipper.  Installed it out of curiosity.Should I try removing it?