Ultimaker/Cura #20523: Model can't be sliced anymore. From perfect G-code to terrible lines and no G-Code terminal to check what is happening.

a4jp-com, 2025-04-19:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Dream Maker Overlord Pro

### Reproduction steps

Adjust setting as liked in Cura.

Slice the model.

Check code in Pronterface. 

Code from Cura: CFFFP_10x10x10.gcode

[image]

Code from Orca Slicer: 10x10x10_PLA_8m23s.gcode

[image]

[Cura-5-10-0-bug-CFFFP_10x10x10.zip](https://github.com/user-attachments/files/19821320/Cura-5-10-0-bug-CFFFP_10x10x10.zip)

The G-Code terminal textarea is also missing making it impossible to tell what is happening on the 

### Actual results

Makes a mess

### Expected results

Everything was good and then suddenly the file couldn't be sliced properly.

I can't even check for errors as Cura is missing the G-code terminal that shows what's happening was we print. 

When I open projects. The material setting say "Update existing" or "Create new". I don't want either option. I want the settings left the way they were most the time. As I tweak settings lots. The files have old settings that sometimes I wanna load.

### Add your .zip and screenshots here ⬇️

[project-file.zip](https://github.com/user-attachments/files/19821371/project-file.zip)

The file looks perfect in Cura but it doesn't slice properly. Please show me how to view the G-Code in Cura like every other slicer. I can just see the input box. I need this to check what is happening as the file prints.

GregValiant, 2025-04-19:
Thanks for the report.
In that wonky gcode file you have this in the StartUp Gode:
G1 E15 F200     ;PRIME
G92 E0          ;ZERO THE EXTRUDED LENGTH
G91             ;Switch to relative positioning
G1 X20 F3000    ;Wipe excess filament
G1 E0.5 F200    ;Push out an additional 0.5 mm filament
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
G92 E0
G92 E0
G1 F2700 E-6.5
;LAYER_COUNT:50
;LAYER:0
Do you see anything wrong there?                        The print starts with the printer in "Relative positioning".

This is "as received".
[image]

This one I added "G90" before ";LAYER_COUNT:50" to put the printer into "Absolute Movement" mode.
[image]

In addition - the 3mf file in your "project.zip" file is invalid.  Is it from Bambu Studio?  They use their own 3mf format.

I'll remove the bug label and go ahead and close this as the problem is in your StartUp Gcode which is not a Cura problem.  I hope you understand.

closed 2025-04-19 by GregValiant

a4jp-com, 2025-04-20:
That's not the startup G-Code I entered. This is my startup code.

```
G28             ;HOME
M501            ;Load stored EEPROM settings
M420 S1         ;Enable bed leveling
G92 E0          ;RESET extruder position
G90             ;Absolute positioning
G1 X0 Y-63 Z5 F2500 ;Move to front
; Prime the extruder
G92 E0          ;ZERO THE EXTRUDED LENGTH
G1 Z0.3         ;Ensure nozzle close to the bed
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
;M109 S200       ;Set the second hotend (T1) to 210°C and wait
G4 S1           ;Short wait for stability
G1 E15 F200     ;PRIME
G92 E0          ;ZERO THE EXTRUDED LENGTH
G91             ;Switch to relative positioning
G1 X20 F3000    ;Wipe excess filament
G1 E0.5 F200    ;Push out 0.5mm of filament
```

It's not reading my G-Code properly.

[image]

a4jp-com, 2025-04-20:
I think I understand what you mean now. Shouldn't your slicer switch back to 
G90             ;Absolute positioning
by itself before it prints like with other slicers? I can't think of any situation where you would want the sliced item to fully switch to relative mode and stay in that mode. I'll add G90 to the end but I'd still consider this a bug and this isn't happening in any other slicer I'm using. I have a Dream Maker Overlord Pro so I can't use Bamboo Studio.

a4jp-com, 2025-04-20:
The G90 at the end of the code fixed the sliced file. Thank you. 

I think Cura should have sliced and shown the mess though. Could you fix the slice preview and add G90 after the start G-Code so the slicer doesn't have this problem. People can add G90 to the extruder is they do want crazy cuts. There should also be a warning we need G90 before printing if we have to set it that way.   

Three other things. 

1. The axis lines are also broken.

[image]

2. And seam lines don't render properly when in random spots and trying to set up a small gap. Isn't there a seam flow setting to reduce to amount of plastic?

3. We can input the G-Code on the monitor screen but not monitor the G-Code. Can you add a box that shows us the values like when we do a bed alignment G33. We need to see the values after calibrating. And it is nice to see the temp on the computer instead of always having to open heated enclosures to check the screen on the printer.

[image]

The G-Code is missing from here ☝

[image]

Shows the G-Code so we can see if there are errors, etc. ☝

The file I sent you as a sample was made with G90 added automatically in Orca Slicer. It is made from lots of different programs like Slic3r, Bamboo Labs and Prusa open source code.

GregValiant, 2025-04-20:
This is the default "StartUp Gcode" from the base printer defintion file "fdmprinter.def.json":

G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0

If your Startup Gcode is anything else then it has been altered by some person or by a definition file override.

How would the slicer know that you didn't want to end up in "Relative Movement" mode?  If I were to manually change my StartUp and some piece of software was to decide "you couldn't possibly want that so I'm changing it to this without telling you" I would not be a happy camper.

The UltiMaker Company does much of the work on Cura.  None of their printers use "USB Printing" as it was found to be too glitchy.  The only reason it is still included with Cura is that there are still UM Original machines out there that require it.
I do not expect UM to pay their people so Cura works better with my Ender 3 Pro.  The same thing applies to your DreamMaker.

I wrote my own piece of software that is specifically for communicating with the printer.  Yes, it has a large textbox that shows the printer responses to everything.  It does not send Gcode files to the printer, but allows 2-way communication, and monitors the USB port for messages during a print.  I think of it as Pronterface on steroids.  I never have to touch the knob on the printer.

[image]