Ultimaker/Cura #20525: Vase mode failing to print some thinner walls on more complex geometry

real-liiva, 2025-04-19:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro selected (mostly custom printer of Theseus, the problem isn't with printing, shouldn't matter)

### Reproduction steps

1. Sliced a custom design designed with vase mode in mind 
2. Gaps were present in part of the print where there shouldn't have been gaps

### Actual results

[image]

[image]

Gap visible at the top of the triangular cutout

### Expected results

There should have been a couple of more layers above where the nozzle's path stopped going across the cutout, at least up until the point where both extrusion lines touch each other and melt together, instead of leaving a gap in the finished print. I would expect the result to be similar to the below screenshot (sliced without spiralized contours, but with 1 wall and no infill)

[image]

### Add your .zip and screenshots here ⬇️

[CE3PRO_pontoon.zip](https://github.com/user-attachments/files/19822167/CE3PRO_pontoon.zip)

[image]
[image]
[image]

GregValiant, 2025-04-19:
Thanks for the report.
That project file kept crashing Cura so I had to import the model and then use my settings.  It sliced fine with Spiralize turned on which doesn't help you but means there probably aren't any errors in the model itself.

[image]

I don't know what's up with the crash-on-open on my system.  I haven't had that problem before.

The Cura team will want your log file for this one.  Maybe there will be a clue in it.  You can use the "Help | Export file for technical support" command and that will create a 3mf with another project file and include your Cura.log file.  Zip that and post the zip folder here.

real-liiva, 2025-04-19:
Here.
[export-20250419-210022.zip](https://github.com/user-attachments/files/19822318/export-20250419-210022.zip)
Weird, i do not experience any crashing on my system, even when opening the project file i sent. Also, i see the gaps i described have occurred even in the screenshot you have attached. I have circled the affected areas in blue in the below screenshot.

[image]

GregValiant, 2025-04-19:
Thank you.
There will always be gaps there because you are in spiralize and there must be an "out and back" move.  If the final stripe was put down then spiralize would have to break because it would be one extrusion to the dead end, and no way back without a simple travel move.

This is in PrusaSlicer and it has the same problem.
[image]

So you have a trig function involving the Layer Height and Line Width and the angle of the walls of the Vee and in the correct combination of LH and LW you may get the Vee to be just close enough together to look like it seals, but it probably won't.
Looking at the model you can see what was done with (what I assume) is the "keel" on top.  That needs to be done with those Vees, or just glue them after printing.

This is Cura in Surface Mode instead of spiralize.  The last single extrusion gets added and the gap is closed.

real-liiva, 2025-04-20:
I know it's probably more a feature request than a bug at this point but Cura supports printing variable wall thickness/variable extrusion width though, shouldn't it be possible to do some variable extrusion width trickery in such situations to guarantee the gap gets closed in any case?

GregValiant, 2025-04-20:
By making an adjustment to a Vee model, it will be closed up in Spiralize.  The part sticking up would be 2 x Line Width wide.
[image]

But that can't be done with the pontoon model because it is a solid.  Unless the model was redesigned to be like an airplane wing or fuselage with a specific wall thickness that dictates the toolpath, you can't close that Vee.  The math of the geometry just won't allow it.

Although it's unfortunate that the model isn't printing "as expected", I just don't see a bug here and I'm inclined to remove the bug label and close this.

real-liiva, 2025-04-22:
[image]

[image]

The issue still occurs even when the model is explicitly hollowed and such a vertical section is added

GregValiant, 2026-09-08:
I'll close this.

closed 2026-09-08 by GregValiant