Ultimaker/Cura #20527: Z access high  in cura 5.9 /5.10 the z access is dset for my printer to 40 mm but it wont let me use more than about 200mm

snoopy9295, 2025-04-19:
### Cura Version

5.9.0/5.10.0

### Operating System

windows 11

### Printer

Creality CR-10 Smart on Klipper

### Reproduction steps

set the extruder to match my printer settings x 300mm y 300mm z  400mm

### Actual results

cant print anything tall 

### Expected results

to be able to print to hight 

### Add your .zip and screenshots here ⬇️

this is driving me up the wall

snoopy9295, 2025-04-19:
need help to fix this

GregValiant, 2025-04-19:
Thanks for the report.

- In Cura, go to "Manage Printers" and then "Machine Settings" and make sure the "Z(Height)" is set correctly.
- While you are there take a look at "Gantry Height" and it should be something reasonable (mine is 25mm which is the distance from the bottom of the nozzle, to the underside of the X beam).
- In the Cura settings go to "Special Modes" and make sure "Print Sequence" is set to "All at Once".  (When you print in "One at a Time" then the maximum Z is limited to the Gantry Height.)

Add a model to the Cura build plate and scale it up so it is 395mm tall.  If it will slice then Cura isn't the problem.  You can save the Gcode and open it in a text editor.  At the beginning will be the minimum and maximum sizes of the print.

Here you can see the line ";MAXZ:395"

[image]

@Asterchades I'm having a senior moment but I seem to recall another function/setting that can limit the Z.  DO you recall?

@snoopy9295 it is also possible that something in your Klipper setup isn't right and that it is limiting the Z movement.

Asterchades, 2025-04-20:
If there is another way to limit it, I've no idea what it might be. Those are the only two settings I can think of as well, and the only circumstance in which the gantry height is relevant as well. Even using the _Z Offset_ plugin doesn't seem to impact it, and the only exclusion features I know of are for horizontal restrictions.

HellAholic, 2025-04-20:
Material shrinkage factor setting also affects it but to a lesser extent.

snoopy9295, 2025-04-20:
Hi Greg
checked the settings for gantry height it is 45 mm and tried the model sizing but it go`s out of range on height and will not allow slicing of the model.

________________________________
From: GregValiant
Sent: Saturday, April 19, 2025 22:19
To: Ultimaker/Cura
Cc: snoopy9295; Mention
Subject: Re: [Ultimaker/Cura] Z access high in cura 5.9 /5.10 the z access is dset for my printer to 40 mm but it wont let me use more than about 200mm (Issue #20527)

[https://avatars.githubusercontent.com/u/64202104?s=20&v=4]GregValiant left a comment (Ultimaker/Cura#20527)<https://github.com/Ultimaker/Cura/issues/20527#issuecomment-2816870118>

Thanks for the report.

  *   In Cura, go to "Manage Printers" and then "Machine Settings" and make sure the "Z(Height)" is set correctly.
  *   While you are there take a look at "Gantry Height" and it should be something reasonable (mine is 25mm which is the distance from the bottom of the nozzle, to the underside of the X beam).
  *   In the Cura settings go to "Special Modes" and make sure "Print Sequence" is set to "All at Once". (When you print in "One at a Time" then the maximum Z is limited to the Gantry Height.)

Add a model to the Cura build plate and scale it up so it is 395mm tall. If it will slice then Cura isn't the problem. You can save the Gcode and open it in a text editor. At the beginning will be the minimum and maximum sizes of the print.

Here you can see the line ";MAXZ:395"

image.png (view on web)<https://github.com/user-attachments/assets/91ca3658-e64d-40f7-9cea-4952f3ffaa5b>

@Asterchades<https://github.com/Asterchades> I'm having a senior moment but I seem to recall another function/setting that can limit the Z. DO you recall?

@snoopy9295<https://github.com/snoopy9295> it is also possible that something in your Klipper setup isn't right and that it is limiting the Z movement.

HellAholic, 2025-04-20:
- Open Cura
- Load a model on your printer (any model would do)
- Click on `Help` on topbar
- Select `Export Package For Technical Support`
[image]
- Select a location and save the file
- Once the success message pops up, click on `Open folder`
[image]
- Right click on the export in the File Explorer (windows 11 file manager)
- Select `Compress To...` > Select `ZIP File`
[image]
- Then drag and drop the zip file in a comment here on github (can't do it by replying on your email)
- Wait a few moments so the file upload completes before pressing the comment button.

GregValiant, 2025-04-20:
I was thinking of "Vertical Scaling Factor Shrinkage Compensation" which (when set to a stupid number) will portray the build volume as being shorter.
This is with it set to 500% and my normal 250 height has been severely reduced.  It still sliced though as the model would still be shorter than 250.
[image]