Ultimaker/Cura #20528: Skew Compensation

BackwardEcho, 2025-04-20:
### Is your feature request related to a problem?

I have a heavily modified Monoprice Mini Delta printer with a fork of Marlin installed instead of the stock firmware. The parts that come off it at really skewed, however due to the limited storage on the controller board Skew Compensation might not fit in the firmware. I would need some other method of Skew Compensation.

I also have a heavily modified Tronxy X5SA printer that runs its stock Chitu firmware. I have yet to know with certainty if the prints coming from that printer are skewed, but I would not have any means for correction as I am certain Skew Compensation isn't supported. I would need some other method of Skew Compensation.

All I can currently do is hardware realignment, but I thought I already aligned both printers to the best of my ability.

### Describe the solution you'd like

Would it be possible to have Skew Compensation tied to slicing? By "Skew Compensation", I am referring to something like Klipper's [Skew Correction](https://www.klipper3d.org/Skew_Correction.html) or Marlin's [Skew Compensation](https://marlinfw.org/docs/gcode/M852.html). I'd imagine having a checkbox to enable or disable compensation, along with the related values you would input, would be found within the printer settings for any given printer. This way the settings are treated like start / end gcodes or the printer's build volume where changing is to be taken more seriously and less likely to be done by accident. As for how the actual compensation would work, I really don't know. I don't know if it would be best to apply a correction to the sliced gcode after it has been made or during the process.

### Describe alternatives you've considered

There is this [Skew Compensation Post Processing Script](https://www.thingiverse.com/thing:3050059), but it hasn't been updated for 6 years. I have no idea if it'd even be compatible with the new(ish) engine Cura 5.0 introduced or how well the post processing will take into account all that the slicer does during the slicing process. The script could very well make the printer attempt to move out of bounds when printing which I feel is something the slicer would catch while the post processing script will not even care for.

### Affected users and/or printers

Anyone that does not have the ability or knowledge to make firmware level changes to their printer. All printers should benefit from this feature, but that would be up the the user to dial in.

GregValiant, 2025-04-20:
Skewing is all about the printer frame and geometry.  If things aren't square and plumb then the prints will be skewed.  I don't see a way have settings to "un-skew" a frame that isn't right.  Most people would never be able to accurately measure the amount of skew and which axis are involved.  Everything would be a three angle compound.  If you cannot precisely measure the amount of skew you won't be able to fix it.

Both "bottom end" and "kit" printers have two problems.  The first is that amateurs are assembling them and they don't pay attention to "skewing".  They want to go as fast as they can so they can start printing.  The other problem with those printers is the single Z drive.  It allows the X beam to sag and if the wheels aren't adjusted just right, then Z-hops are a problem as the X beam gets pushed up but may not want to come right back down when it's time.

Although ABL has made a big difference, it is still possible for a bed and trolley to be out of square to the frame.

A carpenters tri-square and triangle, and a steel or aluminum straight edge should be sufficient for aligning the axis on most open frame printers.  Enclosed units make it harder to get to things, but the idea is the same.  The X, Y, and Z axis must all be at 90° to each other and the Z uprights must be parallel to each other, and the bed must be square to the plane described by the end of the nozzle.  If the printer is picked up and moved, the frame may need to be re-calibrated.

The only way I was able to fix my X beam was to drill a hole in the left Z upright so I could get an allen wrench through and tighten the stupid hidden bolt.  Now it stays rigid - mostly.  When I calibrate the frame (3 or 4 times a year) the X beam is both the first and the last thing I check.  I still don't trust it to stay where I put it.

BackwardEcho, 2025-04-20:
I am aware of all of this. I have rebuilt my Tronxy corexy printer a couple of times and even added metal corner brackets and braces all over while I reassembled it on a glass table (to hopefully keep things flat and square while I tightened bolts). There is only so much anyone can do regarding hardware on a hobby grade machine despite upgrades to the frame and even motion. For being able to measure skew, I have been eyeing Vector 3D's [Calilantern](https://vector3d.shop/products/calilantern-calibration) model, but if I can't apply the compensation values anywhere then I have no reason to purchase it or even test for it.

GregValiant, 2025-04-21:
If you can get the skew compensation values and if all you need is to add M852 to your gcode file, then you could just do it in your startup Gcode.

closed 2025-04-21 by GregValiant

BackwardEcho, 2025-04-21:
[Here is a list of all supported Gcode on my Marlin printer](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/G-M-code-Support) and I just did a test with M852 I0.75 on my Chitu printer, I had it move in a square motion to see if any side would move diagonally with such an extreme value: it did not.

**Neither of them support Skew Compensation in their firmware.** 

So I CANNOT do what you said to do, otherwise I would have done so already.

I am not saying to simply add a M852 line somewhere at the start of the Gcode, I am asking for the slicer to skew the model during the slicing process so that printer's like mine, that do NOT support M852, can compensate for skew.

GregValiant, 2025-04-21:
The UltiMaker Company produces and sells printers.  They will not pay their people to come up with a solution for 3rd party printers that were customized by a 4th party.  It would not make any sense for them to do so.  
You will have to figure another way.

My printer is exact (as far as I can determine) except for the X beam.  It is often off by +/- 0.5mm from left to right.  That is +/- 0.25°.  I cannot see that in my prints, but I do keep an eye on it.

BackwardEcho, 2025-04-21:
So the UltiMaker Company will only add a feature if it primarily benefits their own printers? Even if the feature is something every printer can benefit from, including theirs, just to screw over their competition in the printer hardware market?

BackwardEcho, 2025-04-23:
Apparently so, which is disappointing as I liked this slicer. Guess I'll be going elsewhere. It could have been a cool feature to draw people into the slicer with, but I guess you hate the idea of being an innovator in the market.