Ultimaker/Cura #20529: Slicing failed after model fixed by meshtools.  Cura 5.10.0

glbryant, 2025-04-20:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Ender 3

### Name abnormal settings

Use mesh tools because model was not water tight

### Describe model location

Model is very large for an Ender 3.  Only fits one way

### Describe your model

First time with this model.  Model is not watertight.  Used mesh tools to fix simple holes, then fix model normals

### Add your .zip here ⬇️

[CE3E3V2_L2-S2.zip](https://github.com/user-attachments/files/19824686/CE3E3V2_L2-S2.zip)

GregValiant, 2025-04-20:
Thanks for the report.
Mesh Tools can check a model and inform you if there are problems and it does have some capability to repair models, but it isn't like using a dedicated model repair tool.  You really need to use either an on-line repair site, or MS 3D Builder to repair models.

All that being said, I didn't have any trouble slicing the project.  This is on my Windows 10 system and Cura 5.10.0.
[image]

If you would post your Cura.log file, maybe it will have a clue in it.

GregValiant, 2025-04-20:
I hate throwing away so much filament.
Here I've split the model to get the roof off, and set "Make Overhangs Printable".
[image]

Then there is the roof with supports turned on and "make overhang printable" turned off.
[image]

glbryant, 2025-04-20:
Here is the cura.log file.  Thanks for your help.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

GregValiant, 2025-04-20:
Github rules are that files must be posted directly to the site.  Via email doesn't work.

glbryant, 2025-04-20:
[cura.log](https://github.com/user-attachments/files/19825931/cura.log)

Here is the log.  Thanks for your help.

GregValiant, 2025-04-20:
There are a lot of errors in the log file.  A lot of those errors have to do with your network setup.  I see your printer is and Ender 3 and for the most part, all of Cura's network capability are aimed at UltiMaker printers.  They can't test with anything else as they don't have access to the hardware.  When you install Cura you should just make it a Plain-Jane install.  You don't need an UltiMaker account to slice.

You also have some extra plugins and post processors installed that were not supplied with Cura.  For the most part, post processors should not be a problem since they don't run until you hit the save button.  Plugins can run early and may be a problem.  That's a reason I suggest a basic installation.  "5axes" stuff should be fine, but he has move on from Cura and it doesn't appear that he is keeping his scripts up.

Someone from the Cura team will take a look.

HellAholic, 2025-04-20:
The cause is the `valve` plugin you have installed, disable the plugin and you should be able to slice normally.
Reason for crash is the protocol which the plugin uses to receive the layer data from the engine, and that has a limit of 4194304 bytes in a single transfer and the size of the message for this model is 4310968 bytes.
You can open an issue on the plugin's developer page if you like.

glbryant, 2025-04-20:
Hi, removing the valve plugin worked and I could slice without errors.  Thanks.

glbryant, 2025-04-20:
> I hate throwing away so much filament. Here I've split the model to get the roof off, and set "Make Overhangs Printable". [image]
> 
> Then there is the roof with supports turned on and "make overhang printable" turned off. [image]

This seems to be the best solution.  Did you split the model in Cura?  I would like to learn how to do this.  Would you give me a short list of how you did it?  Best wishes.

GregValiant, 2025-04-20:
I used Microsoft 3D Builder.  It used to be available in the Microsoft App Store.  For Windows 11 I'm not sure what you could do.
When you split a model remember that it will need to be glued together.  I "subtract" 2.2mm diameter x 4mm long pins from a model and 4mm will be on each side of the cutting plane.  When I split the model each piece will have matching 2.2mm diameter holes in them.  Glue pieces of filament into one piece and you have locating pins for the other half of the model.

(This can be done in Cura but it is tougher to locate parts correctly.)

This is in MS 3D Builder:
1. Decide where you want to split a model.  In this case I picked the top of the walls.  The green thing is the cutting plane
[image]

3. Before splitting I would locate where I want my locating holes and subtract my 2.2mm diameter pins.  For this model there are enough feature differences that the two pieces will locate together with no problem.
4. Here is the model half with the roof missing.
[image]

5. Next I save the file.
6. After saving, I "undo" back to the original model and add a splitting plane again but this time it is rotated upside down.
[image]

7. Then split the model and save the file as "roof.stl" or whatever.

Editing a model can result in errors in the model so I re-open each one in MS 3D Builder and the auto-repair utility will run and fix the models.

Then I can bring them into Cura one at a time and set up each slice.

[image]

[L2-S2.zip](https://github.com/user-attachments/files/19827064/L2-S2.zip)

Being able to edit model files is a really handy skill to learn.
Being able to splice gcode files together is another really handy skill.

I'll go ahead and close this as the problem appears to be fixed with the Valve plugin removed.  I'll drop a note on the developers site.

closed 2025-04-20 by GregValiant