Ultimaker/Cura #20531: sqlite3.OperationalError: unable to open database file

jgilmore, 2025-04-21:
### Cura Version

5.10.0

### Operating System

Linux

### Printer

Tevo Tarantula

### Reproduction steps

This error occurred right after I copied my system to a new filesystem (old zfs pool -> new zfs pool. Files where copied with rsync -avzux old new, for each of several filesystem, including /home) But I also hadn't used it in months, so that may be unrelated.

Under 5.9, I go this error. I attempted to correct it by deleting ~/.config/cura. That didn't work, so I downloaded 5.10.0 appImage, set it to executable and ran it, and got the same error.

Either appImage (5.9 and 5.10.0) give the same error, even just after deleting ~/.config/cura.

If cura is attempting to create a database file somewhere else, it's sure as heck not communicating that very well! I don't know where else to look for a possible permissions issue. Note that there was NO .db file in the previous ~/.config/cura/5.9 directory, so it's possible that it's attempting to create the file elsewhere, but how to figure out where? And if so, that's the bug behind the crash.

This may be a duplicate of the closed bug report [here](https://github.com/Ultimaker/Cura/issues/16302)

### Actual results

crash with unhelpful error log

### Expected results

starting the program.

### Add your .zip and screenshots here ⬇️

System Information:
`Cura version: 5.10.0
Cura language: en_US
OS language: en_us
Platform: Linux-6.8.0-57-generic-x86_64-with-glibc2.39
Qt version: 6.6.0
PyQt version: 6.6.0
OpenGL: Not yet initialized`

Error Traceback:
`Traceback (most recent call last):
  File "cura_app.py", line 243, in <module>
  File "cura/CuraApplication.py", line 881, in run
    super().run()
  File "UM/Qt/QtApplication.py", line 385, in run
    super().run()
  File "UM/Application.py", line 290, in run
    self.startSplashWindowPhase()
  File "cura/CuraApplication.py", line 589, in startSplashWindowPhase
    self._container_registry.loadAllMetadata()
  File "cura/Settings/CuraContainerRegistry.py", line 379, in loadAllMetadata
    super().loadAllMetadata()
  File "UM/Settings/ContainerRegistry.py", line 467, in loadAllMetadata
    cursor = self._getDatabaseConnection().cursor()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "UM/Settings/ContainerRegistry.py", line 385, in _getDatabaseConnection
    self._db_connection = self._createDatabaseFile(db_path)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "UM/Settings/ContainerRegistry.py", line 363, in _createDatabaseFile
    connection = db.Connection(db_path)
                 ^^^^^^^^^^^^^^^^^^^^^^
sqlite3.OperationalError: unable to open database file
`

Logs:

`Thread 0x00007931f16006c0 (most recent call first):
  File "UM/Backend/Backend.py", line 209 in _storeStderrToLogThread
  File "threading.py", line 1010 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x00007931f20006c0 (most recent call first):
  File "UM/Backend/Backend.py", line 189 in _storeOutputToLogThread
  File "threading.py", line 1010 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x000079320a0006c0 (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x000079320aa006c0 (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x000079320b4006c0 (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x000079320be006c0 (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x0000793210e006c0 (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x0000793218a006c0 (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x00007932194006c0 (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Thread 0x0000793219e006c0 (most recent call first):
  File "threading.py", line 355 in wait
  File "threading.py", line 507 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1073 in _bootstrap_inner
  File "threading.py", line 1030 in _bootstrap

Current thread 0x00007932588cb740 (most recent call first):
  File "cura/CrashHandler.py", line 397 in _logInfoWidget
  File "cura/CrashHandler.py", line 182 in _createDialog
  File "cura/CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 181 in exceptHook
`

jgilmore, 2025-04-21:
Unable to get it to tell me where it's trying to create the database file. I unpacked the appimage with --appimage-extract, modified the ./squashfs-root/UM/Settings/ContainerRegistry.py to give a more sensible error (i.e. added "raise ValueError(db_path)" immediately before the line that gives the error) and ran it with ./squashfs-root/AppRun

But to no effect - same error. It doesn't appear to be executing the file I modified? Certainly still runs when I delete that file entirely! And gives the same error, but somehow with a less detailed error log... strange. No idea how to debug further, and I need to go finish mowing the lawn, so that's that for now.

`Traceback (most recent call last):
  File "cura_app.py", line 243, in <module>
  File "cura/CuraApplication.py", line 881, in run
    super().run()
  File "UM/Qt/QtApplication.py", line 385, in run
    super().run()
  File "UM/Application.py", line 290, in run
    self.startSplashWindowPhase()
  File "cura/CuraApplication.py", line 589, in startSplashWindowPhase
    self._container_registry.loadAllMetadata()
  File "cura/Settings/CuraContainerRegistry.py", line 379, in loadAllMetadata
    super().loadAllMetadata()
  File "UM/Settings/ContainerRegistry.py", line 467, in loadAllMetadata
  File "UM/Settings/ContainerRegistry.py", line 385, in _getDatabaseConnection
  File "UM/Settings/ContainerRegistry.py", line 363, in _createDatabaseFile
sqlite3.OperationalError: unable to open database file
`

same line numbers, less context, WTF is going on with this shit? Now I feel like a grumpy old man complaining about newfangled executable formats that are too damned opaque to debug.

smartavionics, 2025-04-21:
I think it creates containers.db under ~/.cache/cura