Ultimaker/Cura #20532: Can't delete Custom Profile

alfaniner, 2025-04-21:
### Cura Version

5.10.0

### Operating System

Windows 11

### Printer

Creality Ender-5

### Reproduction steps

I want to delete a Custom Profile that is obsolete.
Went to Help/Show Configuration Folder.
The alternative is to Search for the quality_changes file in the entire system. 

### Actual results

Nothing. No folder pops up, and no file is found.

### Expected results

A folder should open. Suggestions are to find the quality_changes file within that folder and edit it. That file is not found in a search.

### Add your .zip and screenshots here ⬇️

Nothing to add because nothing shows up.

GregValiant, 2025-04-21:
Thanks for the report.
I had that happen ("Help | Show Configuration Folder" didn't work) with one version of Cura but it was a couple of years ago.

That custom profile should be in:
"C:\Users\_your_user_name_\AppData\Roaming\cura\5.10\quality_changes"
If "AppData" is not visible you will need to alter the Windows File Explorer "Folder Options" so "Hidden Folders" are visible.

A custom profile name will be something like:
"creality_greg3pro_air_cleaner_velocity_stack_threaded.inst.cfg"
The printer definition file I use is "creality_greg3pro" and the rest of the file name should have that obsolete profile name in it.

It is also odd that you cannot get rid of it using the "Manage Profiles" dialog.  That's one of the things it's for.
[image]

I might be the world record holder at removing unwanted profiles.  I open a lot of other peoples project files and end up with a lot of profiles and printers and I need to cull the herd once in a while.

I'll leave the bug label on this.  Let us know if you are able to get rid of it.

alfaniner, 2025-04-22:
I totally missed that option of clicking on the three lines for a menu after selecting a profile in that Preferences section and having the Delete selection pop up. That solves it, thanks!

closed 2025-04-22 by alfaniner