Ultimaker/Cura #20541: Y-extruder (2 in 1) sets the standby temperature of filament(extruder) #2

FlyAAM, 2025-04-24:
### Cura Version

5.10

### Operating System

windows 10 64

### Printer

tronxy x5sa 400 (custom klipper)

### Reproduction steps

1. use two filaments with the same print temperature and different standby temperature, for example print temp 230 and standby temp 200
2. startup code fragment
G1 Z5.0 ; Raise nozzle to prevent scratching of heat bed
G1 X0 Y0 ; Move nozzle to Home before heating
M190 S75.0 ; Set bed temperature and wait
M109 S230.0 T0 ; TRUE         ;Set nozzle temp and wait
;;;;
;PRINT_BED_MESH
;;;;
G92 E0 ; Set Extruder position to zero
M82 ;absolute extrusion mode
G92 E0
G92 E0
G10
;LAYER_COUNT:59
;LAYER:0
M107
M104 T1 S200 ;(WRONG)
M204 S1500

3.  T1 200 causes a printing error becouse t.print 230

### Actual results

print error

### Expected results

print error

### Add your .zip and screenshots here ⬇️

In my opinion, an option is needed: one heater or one nozzle, with a check box that will activate/deactivate this.

GregValiant, 2025-04-24:
Thanks for the report.
As I understand this, you have a "2-in-1-out" hot end is that correct?

If that is true then go to the Cura MarketPlace and load the "Printer Settings" plugin.  Restart Cura.
Two of the settings it includes are "Extruders Share Heater" and "Extruders Share Nozzle".  They will need to be ticked.

[image]

Let me know if that is what you were looking for.

FlyAAM, 2025-04-24:
Hi, thanks for the support. This is definitely what we need! I would like
to suggest moving these two options to the "two extruders" section. Thanks
for the great slicer!

GregValiant, 2025-04-24:
I've never understood why all those settings are hidden unless the plugin is loaded.
I'll change this to a feature request.  It seems to me that a page could be added to the Machine Settings dialog and all these settings could be moved over to it.
Right now the setting "Nozzle Diameter" can be entered in two places, but it only updates one-way.  That is really hard to follow.

smartavionics, 2025-04-24:
> I'll change this to a feature request. It seems to me that a page could be added to the Machine Settings dialog and all these settings could be moved over to it.

Err, I think it used to be like that (or similar) but somewhere along the line, the UM devs decided it wasn't to be. Can't remember their reasons.

GregValiant, 2025-04-24:
I recall that the "Shared Heater" setting was on the page and "Shared Nozzle" came along later.  I think that happened when 5.0 was released with the new QT controls.

It seems to me that when I open up "Machine Settings" that all the Machine Settings should be there.  Having a goodly portion missing and only accessible by installing an off-site (and for a new user - unmentioned) plugin is silly and hard to understand.
I applaud @fieldOfView for the plugin or no one would be able to access those needed settings.

If a user wants to manually adjust "Max Feedrates" and "Max Accel" or use "Firmware Retraction" they are only accessible when Printer Settings is loaded leaving the poor user at the mercy of whoever wrote their definition file(s).

FlyAAM, 2025-04-25:
when i was looking for a solution to this problem i saw on forums mention of checkboxes in printer setting, however i was sure that it refers to kura-setting-printer_setting. i was puzzled not finding this setting. now i know that it is a separate plugin. but to tell the truth i think extruder definition should be visible in printer setting. thanks guys for quick help

LilBub, 2025-04-28:
> when i was looking for a solution to this problem i saw on forums mention of checkboxes in printer setting, however i was sure that it refers to cura-setting-printer_setting. i was puzzled not finding this setting. now i know that it is a separate plugin. but to tell the truth i think extruder definition should be visible in printer setting. thanks guys for quick help

Yeah there are a few plugins that, in my opinion, should be part of native Cura at this point.

Especially things like Machine Settings, the SideBar and a couple others.  It would help resolve a few frequent issues I see here, plus put Cura more on par with what other slicers have (without the need of a plug in)