Ultimaker/Cura #20548: [5.7.1] Cura can no longer load in STL files: "cannot open files of the type of "

wosym, 2025-04-26:
### Cura Version

5.7.1

### Operating System

Linux Arch

### Printer

Not relevant

### Reproduction steps

When drag-and-dropping STL files into Cura, the following message is displayed:
`cannot open files of the type of`
(note that the actual file extension is not even in the message)
Uninstalling and reinstalling fixes the problem for 2 or 3 uses, then the same problem starts to occur again. Even without having restarted the App. Restarting the App does not fix the issue, only a full reinstall does.

It appears the extension of the file gets lost somewhere in the process of loading it.

### Actual results

`cannot open files of the type of`

### Expected results

Load the STL, as it did before.

### Add your .zip and screenshots here ⬇️

See above.

GregValiant, 2025-04-27:
Thanks for the report.
Take a look at #16249 and see if @MariMakes suggestion works for you.

wosym, 2025-05-02:
@GregValiant today I encountered the issue again. I checked the Marketplace and could not find the STL Reader at all! (The STL Writer was present)

I rebooted (which solved the issue, as per the above description) and checked again: it was still missing. 
So I guess something is going wrong here.

GregValiant, 2025-05-02:
After the problem hits, use the "Help | Export file for technical support" command.  Zip that smf file and post the zip folder here.  It will contain a project file, the model, all your settings, and your Cura.Log file.
Maybe something in there will give a clue.

wosym, 2025-05-03:
Here's the requested file:

[export-20250503-104308.txt](https://github.com/user-attachments/files/20022608/export-20250503-104308.txt)

(Note that I had to change the file extension from .3mf to .txt because Github didn't allow the former)

Also, an extra observation: Today I tried loading in a 3MF model instead of an STL. This gave the same behavior.

GregValiant, 2025-05-03:
With Cura running, go to the MarketPlace and select the "gear" icon.  Scroll down to "USB Printing" and "disable" it.
After restarting Cura see if you can load a file.  Do you happen to have a USB device plugged in to the computer (ear buds, headphones, etc.?).  It looks like Cura is trying to connect to a serial device that is not a printer and the routine just keeps looping.

wosym, 2025-05-07:
I just encountered the issue again. 
I disabled the USB printing in the Marketplace, restarted the app, tried to load in an STL, ... same issue.

>  Do you happen to have a USB device plugged in to the computer (ear buds, headphones, etc.?).

Not really. I do have a USB-C docking station connection, as well as a KVM switch, but aside from that nothing weird.

HellAholic, 2025-10-16:
Cura receives binary PNG data (the thumbnail or preview icon from the file manager) instead of the actual file path during a drag-and-drop operation.
This explains:
- The incomplete error: `cannot open files of the type of`
- The “previous file” re-loading behavior (because Cura’s file path buffer isn’t being updated)
- The log containing PNG binary data

The problem only occurs after some time or after the user interacts with the file manager — consistent with desktop environment (DE) drag-data caching issues.

Most Linux file managers send multiple MIME types on drag-and-drop, such as:
- text/uri-list → proper file path(s)
- image/png → thumbnail preview (for icons)
- text/plain → fallback

Normally, applications like Cura pick the text/uri-list entry, but if a file manager or custom Qt theme changes the MIME order or encoding, Cura might receive the PNG thumbnail blob first and treat it as the filename. Which looks like it's the case based on the logs.

This issue is specific to:
- AppImage builds (using their own Qt runtime, isolated from system Qt) - Such as Cura
- Certain Arch-based setups with custom GTK/Qt theme bridges (e.g., kvantum, qt5ct, qt6ct, breeze, etc.)

Try another file manager, or temporarily disable the custom themes and see if the issue resolves.

wosym, 2025-10-16:
Oh wow! This is amazing! I've been bathing in frustration for what feels like ages now, and all I had to do was install a different file manager!

I tried Thunar now and after a brief experiment the problem indeed appears to have vanished! (Was able to drag and drop 4 objects without encountering the issue once. Also didn't have to resort to any workarounds)

>    AppImage builds (using their own Qt runtime, isolated from system Qt) - Such as Cura
>    Certain Arch-based setups with custom GTK/Qt theme bridges (e.g., kvantum, qt5ct, qt6ct, breeze, etc.)

I believe this part of your analysis is incorrect though. I had the issue both with AppImage builds, as well as with binaries installed through the package manager. I do use Arch, but not with a GTK or Qt theme I believe.

In any case, many thanks to you @HellAholic ! You have made my life a lot easier :D 

I will leave it up to you maintainers to decide whether this ticket should be closed now, or if a solution should be added now that we know the root cause. (E.g. add a fallback where Cura will try the second MIME type when it detects the first one didn't contain the path to an STL)

wosym, 2025-10-26:
Small update: I now also had the bug with Thunar... So maybe I was just lucky last time...