Ultimaker/Cura #20553: Measurment tool

Avtheaviator, 2025-04-27:
### Is your feature request related to a problem?

When scaling a part I don't know what the center inside dimensions will be because things scale. 

### Describe the solution you'd like

A measuring tool not just a scaling tool of the over all dimension that way when making clips to mount things you can measure the inside to ensure your part is correct. 

### Describe alternatives you've considered

Guessing or printing multiple parts till it's correct

### Affected users and/or printers

EVERYONE

Asterchades, 2025-04-28:
I believe there's already a plugin on the Marketplace which can do this, appropriately enough called "Measure Tool".

fieldOfView, 2025-04-28:
See also https://github.com/fieldOfView/Cura-MeasureTool

GregValiant, 2025-04-28:
@Avtheaviator it seems that @fieldOfView 's plugin is just what you needed.  Let me know if this can be closed.

Avtheaviator, 2025-05-19:
Yes close it. Thank ya guys I'm not the most tech savy.

closed 2025-05-19 by fieldOfView