Ultimaker/Cura #20567: Initial Layer Height error

TheOliverNielsen, 2025-05-01:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Anycubic Kobra Neo

### Name abnormal settings

The slicing error appears when I set Initial Layer Height = 0.1 mm.

### Describe model location

Moving or rotating the model still results in a slicing error.

### Describe your model

If I set Initial Layer Height = 0.2 mm I can successfully slice the model.

### Add your .zip here ⬇️

Setting Initial Layer Height = 0.1 mm works on other models. This is the only model so far that has caused the error.

[CFFFP_1_samla_drybox_base.zip](https://github.com/user-attachments/files/20001582/CFFFP_1_samla_drybox_base.zip)

[image]

GregValiant, 2025-05-02:
Thanks for the report.
That's a bad model.
Cura paints surfaces Red when they are calculated to require support.  Red surfaces facing upward indicates inverted surfaces (the surfaces are facing inside-out).  Cura is recognizing those surfaces as requiring support.  
[image]

Looking up from the bottom - it looks like maybe a part was mirrored and then those two pieces were badly merged.  Cura is thinking like this might be two models.
[image]

It did slice for me but you can see in the lower portion of the model that the "merge" was not done correctly.
[image]

The model needs repair.  If it was to slice for you, you could not trust that slice to print correctly.
I'll remove the bug label from this and close it as it doesn't appear to be a Cura problem.  Let me know if you feel that is premature.

closed 2025-05-02 by GregValiant

TheOliverNielsen, 2025-05-02:
Thank you for the explanation!