Ultimaker/Cura #20568: Ghost image that prints

engpete, 2025-05-02:
### Cura Version

5.10.0

### Operating System

Mac OS Sequoia 15.4.1

### Printer

Ultimaker S3

### Reproduction steps

When dropping an stl model in I get a grey ghost image that actually prints. Why is this happening?

### Actual results

When dropping an stl model in I get a grey ghost image that actually prints. Why is this happening?

### Expected results

No ghost image that prints

### Add your .zip and screenshots here ⬇️

[image]
[image]

Asterchades, 2025-05-02:
If you mean the circle at the back, that's a *Prime Tower* - used to ensure multiple nozzles are flowing correctly when changing back and forth between them. You can adjust its configuration, including enabling or disabling it, in the *Dual Extrusion* settings.

engpete, 2025-05-02:
> GregValiant
>  left a comment
> (Ultimaker/Cura#20568)
> Thanks for the report.
> I agree this looks to be expected behavior given your dual-extruder printer, and that model.
> It looks like the spoon consists of a main part and this lighter yellow inset.
>
> image.png (view on web) <https://github.com/user-attachments/assets/993a236d-c80d-44f7-a697-cd01f85b0cd3>
> If you were to
>
> Set both parts of the model to Extruder 1
> Set the "Skirt and Brim Extruder" to Extruder 1
> Set the "Support Extruder" to Extruder 1
> You could turn off the Prime Tower because Extruder 2 would never be called.
> If you need Extruder 2 for any of those three reasons then you probably need the prime tower, but it is certainly possible to get by without it.
>
> Let us know if turning off the "Prime Tower" addresses the problem. In the mean time I'll remove the bug label.

Hi guys,

Yes that did it. Thanks a lot for your help.

Pete

GregValiant, 2025-05-03:
Very good.  I'll go ahead and close this.

closed 2025-05-03 by GregValiant