Ultimaker/Cura #20571: No nozzle size in cura 5.10 for FLSUN V400

DOD-Xarian, 2025-05-03:
### Cura Version

5.10

### Operating System

Windows 11

### Printer

FLSUN V400

### Reproduction steps

Reinstall Cura several time

### Actual results

I cannot slice

### Expected results

I should be able to select a nozzle size and to slice

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2025-05-03:
Thanks for the report.
The definition file for your FLSun v400 came directly from FLSun and they did not add "variants" for nozzle sizes.
You would need to get with them if you want them added.  Each nozzle size requires it's own "variant" file.

Because there are no variants, there is no "dropdown" box where you can select a nozzle size.  You can make adjustments to the nozzle size in the "Manage Printers", "Machine Settings" on the Extruder tab.   You would need to double check the Line Width settings to make sure they are compatible with the nozzle you have in the machine.

This isn't a Cura bug so I'll remove the label and close this.  I hope you understand.

closed 2025-05-03 by GregValiant

fieldOfView, 2025-05-03:
Did you see the screenshot though? If there are no variants, the printer definition should say there are no variants, and the Nozzle Size dropdown should not show.

GregValiant, 2025-05-03:
From the v400 definition file:
 "metadata":
    {
        "visible": true,
        "author": "FLSUN, Guislain Cyril",
        "manufacturer": "Flsun",
        "file_formats": "text/x-gcode",
        "platform": "flsun_v400.stl",
        "has_machine_quality": true,
        "has_materials": true,
        "machine_extruder_trains": { "0": "flsun_v400_extruder_0" },
        "platform_offset": [
            0,
            -40.6455,
            -43.5
        ]
    },
@DOD-Xarian did you use the FLSun definition or did you install the printer as a "Custom FFF"?