Ultimaker/Cura #20573: Can you add a link to a compiled nightly build?

a4jp-com, 2025-05-04:
### Is your feature request related to a problem?

Is there a way to automate the compiling? I'd love to be able to just download a working exe file to install and test the latest version of Cura. There have been so many good updates but we only have 5.10 now.

### Describe the solution you'd like

Links on the site or Github page to an official compiled build and a nightly build.

### Describe alternatives you've considered

Orca as they have this and it's easy to test.

### Affected users and/or printers

Everyone as it's easier to test and install.

HellAholic, 2025-05-15:
Thanks for the suggestion — we understand the appeal of having easy access to compiled nightly builds of Cura.

However, we intentionally don’t offer public nightly .exe files, and here’s why:
Nightly builds are unstable by nature. They often contain incomplete features, experimental code, and known issues. Making them publicly downloadable invites confusion, bug reports about known problems, and worst of all — unintended use by users who may not understand the risks.

Even with disclaimers, history has shown that many users will treat a nightly build like a production-ready release. This creates unnecessary support overhead and noise, distracting from actual development and proper testing.

While other projects might get away with this due to a smaller user base or simpler scope, Cura supports a huge variety of machines and users — many of whom depend on consistent, predictable behavior.

That said, if you're comfortable building from source, we provide instructions for setting up Cura locally. This gives advanced users access to the latest changes without exposing the wider community to unstable code.

We’re always looking for better ways to involve testers and contributors, but public nightly binaries aren’t the right solution for us.

closed 2025-05-15 by HellAholic

fieldOfView, 2025-05-16:
How about the build artifacts of the github actions?

Totally incomplete, experimental, unsupported, and the downloads expire within a couple of days. But compiled binaries anyway.

a4jp-com, 2025-05-21:
If you change your mind there will be more testers here in the future. We are happy to test stuff for free if it is improving prints.

Also if it's okay, can you release fixes more often? Like once every 2 months (6 times a year)? That way we get things like this https://github.com/Ultimaker/Cura/commit/603df28c2adea8a47b470adf780419ecba57aff4 and updates from the 132 pull requests lined up if they are nice updates. 

Looking forward to helping out if I can.