Ultimaker/Cura #20576: 2 extruders, when using only 1, 2 work non-stop.

FlyAAM, 2025-05-04:
### Cura Version

5.10

### Operating System

Win 10, 64

### Printer

Tronxy x5sa 400 2e (custom klipper)

### Reproduction steps

I print a model with 1 extruder, I tell cura to use extruder 1, but extruder 2 always spins. I tried to make it inactive, but in this case it seems to me that it just copies extruder 1.

### Actual results

Extruder 2 work nonstop 

### Expected results

I guess there is a hidden setting to disable the inactive extruder. Teach me where to find it((

### Add your .zip and screenshots here ⬇️

I haven't zip

GregValiant, 2025-05-04:
Thanks for the report.
I'm not sure what you mean here.  Cura will often use Extruder 2 for support and for the Skirt/Brim.  Go to those settings and explicitly set them to Extruder 1.

There is a bug in 5.10 that will cause Cura to ignore Extruder 2 if it is the only extruder used, but I have not see a problem with using only Extruder 1.

You have a custom printer running Klipper.  We need a project file and your log file to investigate.  The instructions were on the page you filled out for this bug report.  When you right click on a file in Windows File Explorer, one of the options is "Send To..." and then select "Compressed (zip) folder".

FlyAAM, 2025-05-05:
Thanks for such quick replies, Greg! Here are my files: 1 - extruder#2 is not disabled, 2 - extruder#2 is disabled and k

[all_f.zip](https://github.com/user-attachments/files/20034895/all_f.zip)

lipper log. But you are right, in files 1 and 2 there is no switch to extruder T1, I don't know what the problem is, but it causes inconvenience.

FlyAAM, 2025-05-05:
hi, i answered (with files) in a post on git

С наилучшими пожеланиями, Анатолий

GregValiant, 2025-05-05:
Within any Gcode file, Extruder 1 would be "T0" and Extruder 2 would be "T1".

Looking at the "1.gcode" file, there is never a mention of T1 and so it should be cold and do nothing through the entire print.
Looking at the 2.gcode file, it is the same.  There is no mention of T1 and so it should do nothing.
If you look at line 49 in both files:
M109 S205.0 T0 ; Set nozzle temp and wait
That is setting the temperature of T0, but it is not "Switching" to T0.  I think the printer is starting with T0 only because that is the default extruder.

Klipper is highly customizable and therefore highly personal.  It might be that you have it configured in a way that you need to tell it which extruder will be used.  There is a "replacement pattern" that you might need to use.
Some people have had to put something like this into their StartUp Gcode.

{if initial_extruder_nr == 0}
T0 ; call Extruder 1
{elif initial_extruder_nr == 1}
T1 ; call Extruder 2
{endif}

I don't know if that will work for you or not.  Klipper can vary from what would often be considered "Normal" usage.  You will likely have to rework your StartUp configuration.

Regarding this bug report, there is nothing wrong with the Gcode so I'll remove the bug label.  It is the user's responsibility to ensure that the StartUp works with their machine.  You might find help over on Reddit.

UM Cura doesn't officially support Klipper and I don't know that it ever will.  Klipper is so highly customizable that there is no way to be sure that one person uses the same macro name as the next person.

closed 2025-05-05 by GregValiant