Ultimaker/Cura #20577: [5.10.0]  Slicing failed (macos)

manijak, 2025-05-04:
### Cura Version

5.10.0

### Operating System

macOS Sequoia 15.4.1

### Printer

Flsun V400

### Name abnormal settings

Slicing fails on all stl models

### Describe model location

no

### Describe your model

It`s a single layer square, for first-layers testing. Used it many times before. 

### Add your .zip here ⬇️

I used Cura 5.7 for a long time with my Flsun v400 without issues. I today updated Cura to 5.10.0 and every single .stl model fails to slice. 

[flsunv400_slicingfailure.zip](https://github.com/user-attachments/files/20028330/flsunv400_slicingfailure.zip)

[image]

GregValiant, 2025-05-04:
Thanks for the report.
Go to the Cura Marketplace and click on the "gear" icon.  Scroll down to "USB Printing" and disable it.  Restart Cura and try to slice a file.
Let us know if that worked for you.

If you have USB devices plugged into your computer, and if internally they are configured as serial devices, then Cura assumes they are printers and keeps trying to connect.  If (for example) the USB device is earbuds, then Cura can't connect and gets stuck and can't do anything else.

GregValiant, 2025-05-04:
I don't have a MAC system to check on.  Someone from the Cura team will need to take a look.  The log file is full of USB baud rate lines and that indicates that it is trying to connect to something.

HellAholic, 2025-05-07:
The logs show an issue with the printer instance / definition. There is a setting missing, `speed_wall_0_roofing` which indicates either something went wrong during the upgrade (from previous version), or you installed the new version on top of the previous version, or you're using a custom printer definition.
I do see a lot of exceptions in the logs for the upgrade script failing due to issues with a configuration for the quality files.
```
Exception: Exception in quality upgrade with VersionUpgrade52to53.VersionUpgrade52to53: Traceback (most recent call last):
  File "UM/VersionUpgradeManager.py", line 441, in updateFilesData
    upgrade_step_result = upgrade_step(file_data, file_names_without_extension[file_idx])
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/UltiMaker Cura.app/Contents/Resources/share/cura/plugins/VersionUpgrade/VersionUpgrade52to53/VersionUpgrade52to53.py", line 334, in upgradeInstanceContainer
    parser.read_string(serialized)
  File "configparser.py", line 710, in read_string
  File "configparser.py", line 705, in read_file
  File "configparser.py", line 1074, in _read
configparser.DuplicateOptionError: While reading from '<string>' [line 35]: option 'infill_pattern' in section 'values' already exists
```
This usually should show you an error that informs you about a corrupted configuration, if you've seen the error, then the slice failing is to be expected as you're trying to slice with a corrupt configuration.

You can try to resolve the issues manually if you know where they originate from, or attempt to clear the configuration folder (reset) and start fresh, both paths can fix the slice issue.